Ultimaker/Cura #7895: Cura 4.6.1 wont start on my computer

jsnake1489, 2020-06-08:
**Application version**
4.6.1

**Platform**
attaching DxDiag file

[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4747105/DxDiag.txt)

**Printer**
(Which printer was selected in Cura?)
Creality ender 3 
**Reproduction steps**
opened the program
ran compatibilty mode in Win 8 which didnt work 

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Application Continues to crash on opening

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
application wont even open

**Log file**

[cura.log](https://github.com/Ultimaker/Cura/files/4747099/cura.log)

**Additional information**
(Extra information relevant to the issue.)

jsnake1489, 2020-06-11:
> This looks, so far, like a duplicate of #4427
>
> In your DxDiag.txt I can see that you have two monitors and they operate
> on different refresh frequencies (59 vs 60 Hz). There is a bug in the Qt
> version we're using that causes the application to freeze if the Cura
> screen is loaded on the wrong display.
>
> Could you try opening the file
> C:\Users\<username>\AppData\roaming\cura\4.6\cura.cfg with a text editor,
> and under [general] write the following line:
>
> restore_window_geometry = False
>
> This will reset the position of the window to the default monitor, which
> may work better then.

Awesome that fixed it right up and i'm back to work!

Thank you so much!

Ghostkeeper, 2020-06-15:
Okay, closing as a duplicate of #4427 then.

closed 2020-06-15 by Ghostkeeper