Ultimaker/Cura #7899: Dual extruder disables print sequence option

chepo92, 2020-06-09:
**Application version**
4.6.1

**Platform**
Win 8

**Printer**
Custom printer, dual extruder

**Reproduction steps**
1. Try to change print sequence option it says it is disabled due to number of extruders

**Screenshot(s)**

[image]

**Expected results**
Print sequence enabled
I'm not making a dual print, just want to make a simple one by one print with the second extruder

nallath, 2020-06-09:
This is correct. If there is more than one extruder enabled, the print sequence setting is disabled. If you want to use the second extruder to make a one at a time print, you need to disable the first extruder.

closed 2020-06-09 by nallath

chepo92, 2020-06-09:
> 
> 
> This is correct. If there is more than one extruder enabled, the print sequence setting is disabled. If you want to use the second extruder to make a one at a time print, you need to disable the first extruder.

But how? in material tab the enable checkbox is greyed, so I can't disable it 

[image]

chepo92, 2020-06-09:
Restarted Cura and it is selectable now

BlastCom, 2021-04-08:
I have a question about that, I want to print 2 similar small objects that you can refer to as a business card. 
To save time between print, I want to print once at a time. The first one is "Yellow PLA" and the second one is "Black PLA".
I want to purge the first extruder only before printing the second model in a different color.

It could be the same process as one extruder, but before printing a second one there is a little purge tower or line to get rid of the last filament used. (This could save time for the user)

Here is an image for representation minus the purge.

[image]

Ghostkeeper, 2021-04-15:
@BlastCom Look into writing a custom Extruder Start G-code specifically for the second extruder that does the priming. It'll get triggered when it switches to the other extruder then.

yakoder, 2024-09-23:
Seems like the Cura devs keep conflating "number of extruders" with "number of print heads". From what I've seen, there are quite a few single print head (even single nozzle) printers that have multiple extruders on the market. (Same with "lowering the print bed" when many raise the print head instead.)

Considering Cura isn't truly "community" but Ultimaker-specific, this leaves a bad taste in the mouths of those who do not use UM printers.

This could be alleviated by the devs remembering this in their comms to soften the language they use.

This specific issue could be as simple as an option in the Printer Setup to specify number of print heads (or even number of nozzles).

IMHO "is not really desirable" is a lazy answer. Not being able to use Print Sequence is _less desirable_ than the little time it takes to raise the print head to the max.

The _only_ difference between some dual-extruder/single print-head printers and those with a single extruder, is that Cura would need to move the print head to the Prime Tower location and purge the previous material. Which should be treated as an extra model location during Prep/Slicing.