Ultimaker/Cura #7905: Allow Z hop to be modified per model

nitro2k01, 2020-06-09:
Is there any reason why Z hop when retracted cannot be modified as a per model setting? I can think of the possible ambiguity whether the setting should apply when the start position, the stop position or both are within the selected model, but other than that I can't see a fundamental reason why it would be impossible to modify per model.

The intended use case for me is to prevent collisions with already printed objects when doing multi-color inlays. I would specify a region through (for example) a support blocker geometry where Z hop is enabled below a certain Z height and then disabled above that height where it's not needed, to save time and wear on the Z axis mechanics.

A slightly more advanced solution catering exactly to my use case would be the ability to define Z hop exclusion zones over which rapids are always performed using Z hop along the top of the zone. Although I don't expect that this would be generally useful enough to justify the development time.

An alternative I could consider if the feature request is denied is that I write a G-code postprocessing script to do modify Z hop commands.

Ghostkeeper, 2020-06-11:
The official reason is indeed that it's ambiguous. This can get complex, because travel moves can go from model to the same model or from model to a different model, but also from model to support, from support to prime tower, from prime tower through a model to another model, etc. Take for instance the case of a number of concentric rings, half of which have Z hop enabled and the other half don't, and you need to travel from the innermost ring to the outermost ring. Of course, you could think of rules to resolve these cases. Those will be hard to understand for a user, but in the end the print result could be better.

The unofficial reason is that the [travel move routine](https://github.com/Ultimaker/CuraEngine/blob/08fe61a7d28ddd08603a9d9e3e625d31e8ed8071/src/LayerPlan.h#L480) in CuraEngine is not specific to a model. That piece of code doesn't know what model it's travelling from or to, and we didn't find it worth the additional complexity to make it know that, and do something with it, so far.

That's not to say that won't change in the future though. Considering how rarely such settings are adjusted, and how rarely settings are adjusted per model, we'll defer this issue for a time when this becomes more pressing. We won't give it priority for now.

ciesizdz, 2021-09-23:
Hello, @Ghostkeeper 

Not sure if there's any update on this topic. I was looking for the solution for the multi-color bottom surface using z-hop trick (separate image to multiple files,  each for different color; print about 2 layers of each color with z-hop, changing the filament between prints; then print remaining layers of single color). The issue is I don't need z-hops for higher layers and need to slice the image twice, then manually merge the gcode files, which is not especially convenient. Not to mention that it happened to me to get inconsistent results between z-hop enabled and disabled slice -- not sure if I made something wrong (like clicked something between slicing attempts) or slicing (maybe, more specifically, slicing with that function turned on and off) is not deterministic.

Do you know of any plans to restore this feature? Or can suggest another workflow? The method became quite popular recently, which makes me hope for support for simplifying it in Cura.

Ghostkeeper, 2021-10-01:
> Not to mention that it happened to me to get inconsistent results between z-hop enabled and disabled slice -- not sure if I made something wrong (like clicked something between slicing attempts) or [...]

When making a Z hop, Cura will make a direct travel move rather than trying to avoid hitting the walls. It doesn't need to avoid the walls when it flies overhead. This might be the inconsistency that you see. It could affect the order of printing certain things, e.g. if Infill Travel Optimization is enabled.

We don't currently have it planned in our priority list. But use cases like this do make us reconsider, so it's good that you suggest it. In this case though that use case is not something that Ultimaker can use since Ultimaker's printers have a nozzle lifting system which defeats the purpose of that Z hop trick.