Ultimaker/Cura #7906: Dual Extrusion Strategy and Prime Tower

GalaxAjh, 2020-06-09:
**Application version**
Cura 4.6.1

**Platform**
Mac OSX 10.13.6

**Printer**
Dagoma DicoEasy 200 (2 extruders, 1 nozzle)

**Reproduction steps**
I want to print a piece in 2 colors. I set my parameters. For the first layer, If I decide to chose Extruder 1 for build plate adhesion, the prime tower is generated only with extruder 1. If I choose Extruder 2  with the same other parameters, the prime tower is generated with the 2 extruder. Wich in my opinion is the  better solution to prime/purge the extruder. As you can see, there is only 1 line on the skirt to prime extruder 2 (1st picture)

**Screenshot(s)**
[image]
_Build Plate Adhesion : Extruder 1_
[image]
_Build Plate Adhesion : Extruder 2_

**Reproduction steps**
An other thing is : if a layer ends with extruder 1, at the begin of the next layer, there is a new prime. If the extruder 2 ends the layer, there is no prime on the next layer. This means that for the prime tower, 1 layer on 2 is not fully completed and on the next layer, Extruder 2 extrude on the prime tower on empty layer space

[image]
_Layer 6 finished_
[image]
_Very beginning of layer 7_

**Project file**
In my opinion, the first layer must allow the prime of the 2 extruders whatever the selected extruder for the buid plate adhesion(sometimes the second extruder in not ready to print and must be primed). For the next layers, extruders must be purged one by layer. No need to purge extruder 1 when it finished the layer and begin the next.
[image]
Layer 6 finished
[image]
Very begin of layer 7

**Request**
Is it possible to change the running extruder when the nozzle is on the tower not on the subjet. Sometimes, it is necessary to change the t° of the nozzle and I think i's better to wait the good T° outside the piece

**Log file**
Another issue : The saved file does not save the distribution of the extrudeurs
[DD_couvercle.3mf.zip](https://github.com/Ultimaker/Cura/files/4754654/DD_couvercle.3mf.zip)

**Additional information**
Relevant with issue #7864

Sorry I am not aware of how to fill an issue and english is not my native language

Ghostkeeper, 2020-06-11:
> > As you can see, there is only 1 line on the skirt to prime extruder 2 (1st picture)
> 
> This might be a bug, since the settings specify at least two lines (only works for extruder 1). You can work around this by setting 'skirt/brim minimum length' to a higher value.

It's currently the expected behaviour. The skirt is printed with the Platform Adhesion Extruder, but the other extruders are also primed but only until their "minimum skirt length" has been achieved.
I don't see a reason why it should work like this though. Logically all extruders that are used in the print should prime the same way, whether they are the first to start or not. I think the priming of additional extruders was implemented more with the brim in mind. They are the same algorithm in CuraEngine.

mahtDFR, 2020-06-12:
This is added to the backlog as CURA-7504

rburema, 2020-06-12:
Concerning the '2nd extruder doesn't listen to skin line number', we've made internal ticket CURA-7504, so it'll get picked up at some point.

When it comes to the nozzle heating over the object, the extruder start and extruder end positions should have been set for the machine.

fvrmr, 2022-02-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker, we have decided to remove this bug from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-13 by github-actions[bot]