Ultimaker/Cura #7907: Removing Printer corrupts config

chepo92, 2020-06-09:
**Application version**
4.6.1

**Platform**
Win 8

**Printer**
Anet A8 - Ender 3

**Reproduction steps**
1. Remove printer from machine list
2. Restart Cura
3. Cura will show a warning that configuraion seems to be corrupt (and list corrupted configs., Anet A8)
4. The same happens if you remove another printer 

**Screenshot(s)**
[image]

**Actual results**
The error appears every time cura is started

I don't want to reset my configs or other printers

64Florian, 2020-06-09:
Hello,
In fact those settings are referenced in the "recent settings" as I remember 
You have to edit manually some files (OK this is sad but not too difficult)
-> for sure Cura have to handle this by itself ... (In my point of view) 

But for now, could you follow those steps please :
click on "Help" -> "show configuration folder" -> Your explorer should open a "4.6" folder, then zip this folder and add it in this ticket.

When I 'll get your files I'll try to help you to locate few modification you have to do to be back on clean workspace  (I'll share fix version too if I can)

chepo92, 2020-06-09:
Thanks, 
I erased manually the corresponding machine file (Anet A8) in /machine_instances folder I also had to erase the respective config in /extruders folder

I have had previously this issue (and solved) just I didn't find it reported or in any to-do

> In fact those settings are referenced in the "recent settings" as I remember
> You have to edit manually some files (OK this is sad but not too difficult)
> -> for sure Cura have to handle this by itself ... (In my point of view)

I agree

Ghostkeeper, 2020-06-11:
I'm not able to reproduce this issue so far. I added an Anet A8, rebooted Cura, then deleted the printer (I tested to delete it while it was active and while a different printer was active) and restarted Cura again, but no error message comes up.

Did you maybe run Cura as administrator at some point, and it's unable to delete some of the profiles?

64Florian, 2020-06-11:
Hello @Ghostkeeper,
For my case, (_I work on MacOS_) as I remember, I got this issue after deleting "usless" printer manually in the profile folder (and the "active profile" was one profile  in the deleted list, and this profile has unsaved modifications )

unfortunately I try to reproduce it and I only succeed to be back to the 1st splash screen to add "my 1st printer" .. but when I did it (by mistake) I didn't clean all cfg file and Cura find all of then but not all and was "in the middle" 

"Hopfully" I mad a backup of my old "broken workspace" : 
[4.6 Backup.zip](https://github.com/Ultimaker/Cura/files/4767797/4.6.Backup.zip)
I have unzipped and test on 4.6.1 and I have this screen : 
[image]

It should be quite close to @chepo92 issue 

-> If I add a new printer I'm in the "exact same case" as @chepo92 
[image]

Ghostkeeper, 2020-06-15:
We consider manual modification of those files to be out of scope though. It's completely correct to indicate configuration errors if you made mistakes modifying those files.

fieldOfView, 2020-06-15:
Resetting the whole configuration might be overreacting a bit. Perhaps Cura could offer to only remove the affected files. The plugin "Startup Optimiser", available from the Marketplace, adds a button to ignore the affected files so the "Configuration errors" dialog no longer shows up.

64Florian, 2020-06-15:
@chepo92  , could you confirm you use "proper way"  to remove your printer ? (using manage printer)
Thank you

chepo92, 2020-06-16:
> 
> 
> @chepo92 , could you confirm you use "proper way" to remove your printer ? (using manage printer)
> Thank you

Yes I did use de manage printer

I update the "procedure", just reproduced it again, not everyone does this and not every time but the bug exists (I will also edit the original proc.): 
1. Open one instance of cura
2. Create two  custom machines, default configs are ok, (e.g. name "P1" and "P2")
3. Close cura
4. Open two instances of cura
5. Activate P1 in one instance and P2 in the other. 
6. In P1 instance, open printer manager, and delete P1, then close cura
7. Close P2 instance of cura
8. Start a new instance of cura
9. Then you get the error

sometimes I'm adding and deleting machines and having multiple instances of cura open (3D printer technical service and 3D printing for my own) so I have a window open for every printer I'm working on, and sometimes I delete some uncommon printers to clean up my machine list. 
Again, not everyone does this and not every time, but the bug exists, so I might close this but leave ir for the record if you agree. 

I guess this has to do with the second instance of cura being closed and overwriting or messing with the printer configs that the first instance erased.

Now I got to fix the files as all printers seem to have been removed (It showed the welcome, agreement and add printer window) 

[image]

fieldOfView, 2020-06-16:
Having two instances of Cura open at the same time is going to lead to issues, because they each don't know what the other is changing in the configuration.

64Florian, 2020-06-16:
> Now I got to fix the files as all printers seem to have been removed (It showed the welcome, agreement and add printer window)

in fact add a printer and the other printer should be they too .. then delete using Cura the new profile 
If you have lose all your profile, you have a backup on this page ;-)

chepo92, 2020-06-17:
The files that Cura says are corrupted are located in configs folder, and sub folders: 
/machine_instances: delete the file of the machine that was deleted in printer manager

/extruders: delete all files that cura says are corrupted, eg. custom_extruder_1+%232.extruder.cfg

I don't know why cura created like 9 extruder files that needed to be deleted, it was just a new added printer and didn't customize any option

Ghostkeeper, 2020-06-17:
> Could Cura offer to "drop" those configuration file (by asking user for it) OR clear all may be instead of just offer to erase all ?

We've been asked this before. The problem is that these are (typically) files that have bad references, but other files can have references to these files as well, leading to a cascade of more files needing to be repaired. The user is basically going to get into multiple restart, each peeling down a layer of broken files.

Add-on to that is that it shouldn't happen in the first place, of course. Our efforts are more towards preventing that from happening.

Ghostkeeper, 2020-06-18:
> I don't know why cura created like 9 extruder files that needed to be deleted, it was just a new added printer and didn't customize any option

Cura will create a file for every profile, regardless of whether there are any modifications in the profile. That makes the logic simpler, and each of these empty profiles is just a few hundred bytes. The Custom FFF Printer has 8 extruders and there's a separate profile for the global stack, so 9 makes sense.

nallath, 2020-06-19:
I've fixed an issue where all machines would be saved on every change. This makes the likelihood of this issue occurring significantly lower.