Ultimaker/Cura #7908: Dot displayed but not working as decimal separator

Entretoize, 2020-06-10:
The decimal numbers are displayed with a dot but if you change them you need to use a comma which is really a bug.
For example I tried to change the nozzle size in machine settings, it is displayed 0.4 then I put 0.8 but this setting was never save, after searching I find the [issue 7667](https://github.com/Ultimaker/Cura/issues/7667) and the solution worked, but the real issue is about the comma/dot use and needs to be corrected in cura.

mahtDFR, 2020-06-10:
Duplicate of #7667

closed 2020-06-10 by mahtDFR

Entretoize, 2020-06-10:
Did you read the post ??! Please reopen, this is a bug and the 7667 is about a precise issue that found a workaround but that's not a solution, it needs a code correction.

Ghostkeeper, 2020-06-11:
Yeah #7667 is also a duplicate of https://github.com/Ultimaker/Cura/issues/6546, so please see that one rather. That issue is still open, and indeed needs a code correction.

The numeric text boxes in that dialogue in particular are system-styled and created by the Qt GUI framework, which handles the decimal separator according to your computer's locale. Since you're most likely in a locale that officially uses a comma as decimal separator, you would need to use that one. This is inconsistent with the rest of the interface which allows both comma and period but always changes it into a period, so that needs to be fixed.

Entretoize, 2020-06-11:
Ok, clearer now. Thanks