Ultimaker/Cura #7909: Every print turns out extremely weak and with small holes in it.

InterFelix, 2020-06-10:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 2004 x64 on i7 4770k and GTX 970

**Printer**
(Which printer was selected in Cura?)
Sidewinder X1

**Reproduction steps**
Not sure at all.
Cura worked very well for a few months, but started producing extremely underextruded prints (to the point where you can basically see through them)no matter what profile or settings selected (I even set Flow to 150%, to no effect.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)
Unusable prints

**Expected results**
(What should happen after the above steps have been followed.)
Well, prints that I can actually use, like the ones PrusaSlicer puts out.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
[CFFFP_Brushless-Wellenmitnehmer-v1.zip](https://github.com/Ultimaker/Cura/files/4756559/CFFFP_Brushless-Wellenmitnehmer-v1.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
Since I can print successfully with PrusaSlicer, the printer can be ruled out as a cause.
It's kind of buggy with some of my models though, which is why I want to use Cura instead, since it was less buggy when it was working.

smartavionics, 2020-06-10:
Perhaps it's using a filament diameter of 2.85 rather than 1.75 so everything is underextruded?

InterFelix, 2020-06-10:
Actually, yes. But why?! I definitely didn't change it. Thanks a lot, though!

smartavionics, 2020-06-10:
I don't know why it changed that value, glad it's fixed though, cheers!

closed 2020-06-10 by InterFelix