Ultimaker/Cura #7910: Unnecessary supports

InterFelix, 2020-06-10:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 2004 x64, i7 4770k, GTX 970

**Printer**
(Which printer was selected in Cura?)
Artillery Sidewinder X1

**Reproduction steps**
No particular steps. Some models just show unnecessary supports.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
Well, those pointless supports (that actually support nothing) show up.

**Expected results**
(What should happen after the above steps have been followed.)
Only generate suports that actually support the part.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
[CFFFP_Brushless-Wellenmitnehmer-v1.zip](https://github.com/Ultimaker/Cura/files/4756938/CFFFP_Brushless-Wellenmitnehmer-v1.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-06-10:
I could reproduce this in 4.6.1 with the project file provided. It seems to be specific to the Artillery Sidewinder X1 profile, because it wouldn't reproduce with the Ultimaker S5 profile.

InterFelix, 2020-06-10:
Now that's curious. I can't think of a setting that could cause this behaviour (at least directly). Maybe some setting that triggers a bug.

mahtDFR, 2020-06-10:
As a workaround, do this:

[image]

InterFelix, 2020-06-10:
> 
> 
> I could reproduce this in 4.6.1 with the project file provided. It seems to be specific to the Artillery Sidewinder X1 profile, because it wouldn't reproduce with the Ultimaker S5 profile. The Artillery Sidewinder profile doesn't ship with Cura, right? It's a custom add-on?

Yes, It's custom. No idea why this happens, but the workaround seems to work okay.

mahtDFR, 2020-06-10:
As it's a custom profile we won't be putting time into it, so I'll defer this
(I suspect this is a duplicate ticket but can't find it right now)

mahtDFR, 2020-06-10:
We have a sort-of-policy of not fixing custom added profiles (i.e. ones you've downloaded externally) because they're a bit of a nightmare to maintain for. It would be good if someone could contribute a definition for this printer for us as a pull request. Details about this here:
https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura

As far as I understand this issue is a known issue with stair stepping, which is already on our backlog (CURA-6989). That means this is a duplicate of #6379, so I will close this ticket and you can continue the discussion there if necessary.

closed 2020-06-10 by mahtDFR

rburema, 2020-06-25:
We might finally have squashed this bug! When we eventually release 4.7 (which might take a bit longer than usual) you can check it out.