Ultimaker/Cura #7921: [4.6.1] Cura running slow on MacOS Mojave 10.14.6, but even after Cura is closed, the computer is slowed down considerably.

lavik17, 2020-06-11:
**Application version**
4.6.1

**Platform**
MacOS Mojave 10.14.6
Macbook Pro 15", 2.5 GHz Intel Core i7
16 GB 1600 MHz DDR3
GPU: AMD Radeon R9 M370X 2 GB
Intel Iris Pro 1536 MB

**Printer**
Ender3

**Reproduction steps**
1. Open Cura 4.6.1
2.Rotating the bed without importing any object - slow response
3. Trying other software (e.g. Chrome), slow response
4. Closing Cura, still a slow response from other software.

rburema, 2020-06-12:
Could you attach the logs? The template is quite clear on that, and I think they could be of help in this case.

The log can be found in the user folder (see here for details on how to access those) or you can click on 'show configuration folder' in the help menu within Cura itself.

lavik17, 2020-06-14:
I could not recreate the error today, therefore I am closing this bug report.

closed 2020-06-14 by lavik17

rburema, 2020-06-16:
Thanks for closing!