Ultimaker/Cura #7922: Pause at height Redo layer wrong z value

Erik1488, 2020-06-11:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Generic CFFFP (Anet A8)

**Reproduction steps**
1. Choose Pause at height
2. select any heigth/layer
3. set Redo Layers to any number greater zero

**Screenshot(s)**
[image]

**Actual results**
Z is the value after the stop but should be this before the layers

**Expected results**
Z is heighter than expected.

Ghostkeeper, 2020-06-15:
From your screenshot, the Z coordinate there seems to be correct. It's the Z coordinate that the g-code was at before the pause. It's listed on line 22998 in your screenshot. Why would you expect that to go back to Z 1.65? That's a few layers down of where it last printed, so your nozzle would crash into your print.

It should be noted that in the next version, more than 1 redo layer is no longer possible.

Erik1488, 2020-06-16:
Even one redo layer is not possible. Because it goto the next layer and
after this it is go back.

Theoreticly one redo layer is engouth, but practily two work better without
a crash.

The Pause at height script need more option to fuse the layers. Such as
"slow speed for ... layers", "flow" and "restart at heigth".

Ghostkeeper, 2020-06-17:
Why would you expect the Z coordinate to go back to 1.65 though? Your actual results are my expected results.

LvlLord, 2020-06-17:
I have similar issues, but without using redo layers.
For testing the pause function, I created a simple "Nut-test" model which should stop at 2 and 4 mm to insert a nut into the print (here just a test).
I noticed 3 things which are wrong here:
1. When using the "height" feature and using 2 and 4 mm, only one of them are actually working. The other uses a bogus height value (15 mm) and is placed before the 2 mm one (see screenshot). When directly using layer height, they are positioned correctly inside the gcode.
2. The saved X and Y position is wrong. It moves "back" to X3 and Y3, even though the old positions where X195.xx and Y196.xx (and new ones will be in the same area). It will then extrude useless amount of filament while moving from X3 Y3 to the real position.
3. Why does it extrude the set amount of filament and then retract it again? (lines 2259/2260)

Using Cura 4.6.1 on Windows 10 x64 2004 19041.329.

[image]
[image]
[image]

Erik1488, 2020-06-18:
> Why would you expect the Z coordinate to go back to 1.65 though? 
Because i want to redo 2 layers. I dont want to do the redo the second last (or last) layer with Z=2.05.

Ghostkeeper, 2020-06-19:
@LvlLord Your issues are different from Erik's. I think the 15mm is coming from the two post-processing scripts interfering with each other, with the "too close to bed" line. Please create a separate issue though. Your problem has nothing to do with Redo Layers resuming at the wrong Z coordinate. When you do, provide a project file as well. That'll allow us to see what it's doing wrong and reproduce your bug :smiley:

> Because i want to redo 2 layers. I dont want to do the redo the second last (or last) layer with Z=2.05.

Ah that's what confused me. The last layer was printed at Z2.05 and presumably the layer before that at Z1.85. So redoing the last two layers would move it to Z1.85. That's why the Z2.05 was correct in my opinion. With just one layer, it redoes the last layer at the same height as what the last layer was printed at, which was Z2.05.

We did remove this feature though in the upcoming version so there it's no longer an issue. It was removed because indeed moving to Z1.65 would crash your nozzle 0.4mm into your print.

Erik1488, 2020-06-19:
>We did remove this feature though in the upcoming version so there it's no longer an issue. It was >removed because indeed moving to Z1.65 would crash your nozzle 0.4mm into your print.
The nozzle would go 0.2 mm in the print. The current heigth at pause is 1.85 mm.

Ironing after pause and a close layer before pause works good for me. Mabey you can at this insteed of redo layer in the script?

Ghostkeeper, 2020-06-26:
It already printed a layer at 1.85mm and 2.05mm before the pause. Those layers were printed before pausing, so they were extruded normally. It would penetrate both of those layers.

I can see the sense in ironing after the pause. That's basically what the redo layer does, only at 100% flow rather than 10%. What do you mean with "a close layer before pause" though?

Erik1488, 2020-06-28:
I will post the files to explain my Problem and the solution.

Ghostkeeper, 2020-07-02:
Note that attachments to e-mail replies don't get posted here on Github. We can't see them. You'd need to go to [your issue page](https://github.com/Ultimaker/Cura/issues/7922) on Github's website to post your files.

closed 2020-07-16 by no-response[bot]