Ultimaker/Cura #7923: [4.6.1] not generating g-codes for rectration on linux

ad7718, 2020-06-11:
**Application version**
Cura 4.6.1

**Platform**
MX linux(based on Debian buster)

**Printer**
Tevo tarantula with Marlin firmware, using default print profile for generic PETG

**Reproduction steps**
1. Sliced this model https://www.thingiverse.com/thing:909901
2. Opened G-CODE and there is no retractions there

**Screenshots**
Screenshots are not needed, attaching g-code file.

**Actual results**
No retraction commands in g-code, so the extruder motor is only moving forward. 

**Expected results**
Command for retractions in g-code.

[model and project.zip](https://github.com/Ultimaker/Cura/files/4767113/model.and.project.zip)

ad7718, 2020-06-11:
Hm... Seems that it generates codes that moves E backwards, but motor was not rotating in opposite direction. But I had no such problem with default settings on prusaSlicer.

rburema, 2020-06-12:
No `G10` so, `machine_firmware_retract` must be off for that machine, causing our engine to retract via `G1` instead. Next, take a look at lines 935, 956 and 968 of the gcode you attached. Relative indeed. Which you've already noticed yourself in the second comment.

What I see in see in the printer definition is that the printer uses RepRap gcode flavour:
```
"machine_gcode_flavor": { "default_value": "RepRap (Marlin/Sprinter)" },
```
No value for relative extrusion (or not) is set in the printer definition itself.

When I look at the `relative_extrusion` value in the 'defaults unless specified otherwise', I see this as value for `relative_extrusion`:
```
"default_value": false,
"value": "machine_gcode_flavor==\"RepRap (RepRap)\"",
```
(Notice the difference between the part in the brackets.)

That means, if you want relative extrusion, you should check the 'Relative Extrusion' checkbox in the settings. If that's supposed to be the default for this type of printer, please either make a PR, or, since this would be a small change, indicate that in a response.

ad7718, 2020-06-12:
Yes, I know that I can check 'RelativeExtrusion' checkbox in Cura, but I think printer must work in both modes, I'll need to do few test to check if it is a firmware bug.

rburema, 2020-06-12:
Please do, when it comes to physical machines, we're (mostly) limited to Ultimaker ones (for somewhat obvious reasons). If there's a bug in 3rd party firmware, there isn't much we can do about it.

ad7718, 2020-06-13:
Don't know what happened, it prints in both modes now. I'll reopen this if I'll find what was wrong. Thanks for attention!

closed 2020-06-13 by ad7718

rburema, 2020-06-13:
No problem!