Ultimaker/Cura #7924: All versions of Cura crash on newly built computer system.

mainstreet3d, 2020-06-11:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mainstreet3d, 2020-06-11:
I am still having trouble with Cura crashing , I have tried different versions with no luck . I am using a new computer with a SSD 1TB , 16gb ram , i5 Intel cpu .

Windows 10 Pro , any ideas ? I have tried running it as admin, but no luck.

rburema, 2020-06-12:
Could you _please_ just fill out the template. Specifically, any sort of logs might help.

The log can be found in the user folder (see [here](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) for details on how to access those) or you can click on 'show configuration folder' in the help menu within Cura itself.

Also, what message do you see when it's crashing? Or does it crash before doing anything?

ZetterSploit, 2020-06-12:
no longer works on windows and our 22 computers ... latest version 4.6

rburema, 2020-06-12:
@ZetterSploit  Then for you too, _please_ attach the relevant information, like the _logs_ I just mentioned. We can't telepathically check what's wrong, we need some info to even begin digging.

ZetterSploit, 2020-06-12:
> @ZetterSploit  Then for you too, _please_ attach the relevant information, like the _logs_ I just mentioned. We can't telepathically check what's wrong, we need some info to even begin digging.

I just point out that this bug on the last update of windows, whether you fix it or not it doesn't matter to me.  Just to be nice;).  telepathy?  lol!  good luck ;)

Ghostkeeper, 2020-06-15:
Still runs fine for me on Windows. We're going to need a log file to begin debugging what went wrong on your side.

closed 2020-06-30 by no-response[bot]