Ultimaker/Cura #7925: min_skin_width_for_expansion has bogus default_value

richfelker, 2020-06-11:
The `default_value` for `min_skin_width_for_expansion` is 2.24. This seems to have no basis or relation to the `value` expression, and causes CuraEngine used without the GUI to introduce huge gaps into the skin in odd places, due to preshrink being applied but corresponding expansion being skipped. The `default_value` that would correspond to the `value` expression is 0, and I believe it should be set to either 0 or some very small value close to 0.

rburema, 2020-06-12:
Hm, I see the default value for `max_skin_angle_for_expansion` was updated at some point from 20 to 90, but the default of 20 was set at the same time as the default for 2.24 for the then somewhat new child setting (like you said `min_skin_width_for_expansion`) wasn't updated.

rburema, 2020-06-12:
Fixed in new commit :-)

closed 2020-06-12 by rburema

richfelker, 2020-06-12:
Interesting. Somehow I can't find the change from 20 to 90 at all in the git log. It goes from one diff context where there's a 20 to the next where there's a 90 with no change in between changing it from 20 to 90. Must be some awfulness of repeatedly merging branches back and forth from each other and git log missing the relevant commit...

richfelker, 2020-06-12:
OK, the functional change seems to have been hidden in a completely unrelated merge commit 245796eb08. 🤦

rburema, 2020-06-12:
I like to think we're getting a bit better about that sort of thing ... but we've not really cared much about the history looking clean until recently. (Even then clean is relative of course.)

commit 29e68fe4 (2020-06-12): Fix default value min_skin_width_for_expansion.

At some point max_skin_angle_for_expansion s deafult value was updated, but the default value for its child, min_skin_width_for_expansion wasn't altered to reflect that. Fixes #7925