Ultimaker/Cura #7926: unreadable text in "Discard or Keep changes" window

GalaxAjh, 2020-06-11:
**Application version**
4.6.1

**Platform**
Mac OSX 10.13.6

**Printer**
Dagoma DiscoEasy 200 (dual extruder)

**Reproduction steps**
Changing the material in tab (top of screen in window "prepare)

**Screenshot(s)**
[image]
[image]

**Actual results**
The text in unreadable

**Expected results**
Readable text. Eventually, to resize the window 

**Additional information**
ScreenShots are in english and in french

V.

rburema, 2020-06-12:
What is your screen resolution? Do you apply any screen-scaling?

GalaxAjh, 2020-06-12:
No it is just a part of the screen. My monitor is an old Apple 30' display at the native résolution
juste below, a screen shot of the full display. 
[image]
[image]

rburema, 2020-06-12:
I did some investigation, and it seems this (sub)window doesn't listen to the width/height we get from the theme, but stays the same dimensions no matter what value is put in there.

I'll bring it up with the rest of the team next week.

Ghostkeeper, 2020-06-15:
I've made three changes:
* I've fixed getting the width/height from the theme.
* I've allowed resizing that dialogue.
* I've turned on eliding of the setting names. So if the setting names are too long they'll be cut off with an ellipsis. This way the text remains readable.

commit 09298eac (2020-06-15): Elide setting name if it's too long

This keeps the text readable.

Fixes #7926.
closed 2020-06-15 by Ghostkeeper