Ultimaker/Cura #7927: Translation error -> French

GalaxAjh, 2020-06-11:
**Application version**
4.6.1

**Platform**
Mac OSX 10.13.6

**Printer**
Dagoma DiscoEasy 200

**Reproduction steps**
Translation error 
On the translation, it is written "_Tour primaire_" witch means "_Primary Tower_"
The good text for "_Prime Tower_"must be "_Tour d'amorçage_"
Prime -> Amorçage

**Screenshot(s)**
[image]
[image]

**Actual results**
You can find this in tabs Material and Dual Extrusion. Probably in other tabs but I do not use for the moment

V.

fieldOfView, 2020-06-12:
I think this was recently fixed (for Cura 4.7) here: https://github.com/Ultimaker/Cura/pull/7838

mahtDFR, 2020-06-12:
Fixed indeed, will close this off now.

closed 2020-06-12 by mahtDFR