Ultimaker/Cura #7929: 4.6.1 support issues

realacc, 2020-06-12:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-06-15:
Well done for posting the issue template. Next step: fill it in so we can try to help you.

rburema, 2020-06-16:
Found a brief description in the comments of the template. Edited the original to show it. Still not much to go on.

@realacc Have you tried adjusting 'Support Z Distance', 'Support X/Y Distance' and/or 'Minimum Support X/Y Distance'?

Alternatively, you could attach a screenshot or even the project, so we can see what's going on.

Thirdly, we've seen some support issues due to stair stepping not doing what's expected. A bug that we're working on and which will probably be fixed in 4.7. To test if you're affected, set either 'Support Stair Step Height' or 'Support Stair Step Minimum Width' to 0.

realacc, 2020-06-16:
Hello, thanks for the response back. Yup I did try changing the support x/y
distance from .8 to .3 and it worked. I will also try the stair stepper
height to 0. Thanks again for your help.

HeathHicks, 2020-06-23:
Is there a list of command line prompts and custom parameters for the silent deployment and configuration for either the exe or the msi?

rburema, 2020-06-24:
@HeathHicks Is this what you meant? : https://stackoverflow.com/questions/8560166/silent-installation-of-a-msi-package/8560308#:~:text=You%20should%20be%20able%20to,end%20of%20the%20msiexec%20parameters.&text=You%20can%20read%20the%20options,options%20from%20Start%20%2D%3E%20Run.

closed 2020-06-30 by no-response[bot]