Ultimaker/Cura #7931: [4.6.1] Support starts printing in midair

louiswouters, 2020-06-14:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. Load up the attached project.
2. Slice the model

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
The inner support slices only a small section for the first 3 layers

**Expected results**
The first 3 support layers should be identical to the ones on top to prevent printing support in midair

**Project file**
[SupportSliceBug.zip](https://github.com/Ultimaker/Cura/files/4776499/SupportSliceBug.zip)

**Log file**
not relevant

**Additional information**
none

mahtDFR, 2020-06-15:
I was able to reproduce this. Two things came to mind: the model wasn't 100% flat, and a known issue with stair stepping.

I used the lay flat option in Cura, but that didn't change anything.

However, setting stair step options to 0 did produce better (but not perfect) results.

[image]

So it seems this is a duplicate of #6379. I'll close this ticket off and you can continue the discussion there if needed.

closed 2020-06-15 by mahtDFR

rburema, 2020-06-25:
We might finally have squashed this bug! When we eventually release 4.7 (which might take a bit longer than usual) you can check it out.