Ultimaker/Cura #7932: Dual extrusion project is saved but all pieces have the save extruder

GalaxAjh, 2020-06-14:
**Application version**
4.6.1

**Platform**
Mac OSX 10.13.6 

**Printer**
Dagoma Disco Easy 200 (dual extruder)

**Reproduction steps**
setting what part is printed with what extruder
Saving the project.
Close Cture
Open Cura
Re-open the project
All parts are setted to be printed with extruder 1

**Screenshot(s)**
[image]
Project saved
[image]
Project re-opened

**Actual results**
All parts are setted to be printed with extruder 1

**Expected results**
Re-openning as the project was closed

**Project file**

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4777257/cura.log)

**Additional information**
No additionnal information.

mahtDFR, 2020-06-15:
Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it with the same settings as you.

mahtDFR, 2020-06-15:
I was unable to reproduce this.

OS: Microsoft Windows 10 Pro 10.0.18363 Build 18363
Cura 4.6.1

Here's what I did:

Print two cubes with different extruders
Save project file (3mf)
Close project
Close Cura
Reopen Cura
Open project file

Assigned dual extrusion nozzles are preserved as expected.

[image]

Perhaps it's Mac specific? Anyone able to test this on a Mac? @GalaxAjh perhaps you can share your project file here for analysis.

GalaxAjh, 2020-06-15:
[DD_test_sav.3mf.zip](https://github.com/Ultimaker/Cura/files/4779739/DD_test_sav.3mf.zip)

Here it is.
Not sure it will help, if the problem occurs during the saving.

V.

mahtDFR, 2020-06-15:
Seems to be loading fine for me in 4.6.1. So the project is saving just fine it seems. How strange!

[image]

mahtDFR, 2020-06-15:
Your problem is not reproducible for us, as opening in 4.6.1 seems to work fine on Mac and Windows.

However we just tried this in master (Mac and Windows) and your project file caused a crash.

Windows traceback:

`Traceback (most recent call last):
  File "C:\dev\Cura\cura\CuraApplication.py", line 1057, in event
    return super().event(event)
  File "C:\dev\Uranium\UM\Qt\QtApplication.py", line 464, in event
    event._function_event.call()
  File "C:\dev\Uranium\UM\Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "C:\dev\Cura\cura\Utils\Threading.py", line 34, in _handle_call
    ico.result = func(*args, **kwargs)
  File "C:\dev\Cura\plugins\3MFReader\ThreeMFWorkspaceReader.py", line 731, in read
    self._processQualityChanges(global_stack)
  File "C:\dev\Cura\plugins\3MFReader\ThreeMFWorkspaceReader.py", line 810, in _processQualityChanges
    extruder_stack = global_stack.extruderList[int(position)]
IndexError: list index out of range`

Windows log:

` Thread 0x000038c4 (most recent call first):
  File "C:\dev\Uranium\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 862 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x00000f20 (most recent call first):
  File "C:\dev\Uranium\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 862 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x000029d0 (most recent call first):
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 297 in wait
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf\__init__.py", line 2117 in wait
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf\__init__.py", line 1434 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x000035b8 (most recent call first):
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 297 in wait
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 549 in wait
  File "C:\dev\Cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 862 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x000023a8 (most recent call first):
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 297 in wait
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf\__init__.py", line 2117 in wait
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf\__init__.py", line 1262 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x00001d44 (most recent call first):
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\site-packages\zeroconf\__init__.py", line 1183 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x0000086c (most recent call first):
  File "C:\dev\Cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 862 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x00003434 (most recent call first):
  File "C:\dev\Cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
[... 53 lines cut ...]
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 293 in wait
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 424 in acquire
  File "C:\dev\Uranium\UM\JobQueue.py", line 98 in _nextJob
  File "C:\dev\Uranium\UM\JobQueue.py", line 123 in run
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Thread 0x00004148 (most recent call first):
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 293 in wait
[... cut ...]
  File "C:\Users\m.jani\AppData\Local\Programs\Python\Python35\lib\threading.py", line 882 in _bootstrap

Current thread 0x00000250 (most recent call first):
  File "C:\dev\Cura\cura\CrashHandler.py", line 381 in _logInfoWidget
  File "C:\dev\Cura\cura\CrashHandler.py", line 182 in _createDialog
  File "C:\dev\Cura\cura\CrashHandler.py", line 86 in __init__
  File "C:/dev/Cura/cura_app.py", line 166 in exceptHook
  File "C:\dev\Uranium\UM\Qt\QtApplication.py", line 378 in exec_
  File "C:\dev\Cura\cura\CuraApplication.py", line 841 in run
  File "C:/dev/Cura/cura_app.py", line 234 in <module> `

GalaxAjh, 2020-06-15:
Here is my configuration folder
[4.6.zip](https://github.com/Ultimaker/Cura/files/4782753/4.6.zip)

GalaxAjh, 2020-06-15:
I confirm that the problem comes at the opening on the (my) Mac. Trying to open this Mac project on a PC Win10 : ok
Create a new project on the PC (dual extrusion), saving it and opening it on my Mac and the whole project is set on extruder 1.
I try to set one or more pieces on extruder 2, saving. At the re-opening, all is set on extruder 1
V.

fieldOfView, 2020-06-16:
In your General preferences, under "Opening and saving files", what do you have selected as the "Default behavior when opening a project file"?

mahtDFR, 2020-06-16:
@GalaxAjh I've just loaded you configuration and I see that you have 'load as models' instead of 'load as project' enabled. Can you check this please?

GalaxAjh, 2020-06-16:
Hi
That does the trick. In fact, that was the default option and I do not touch it.
Thanks. I think you can close now this issue.

closed 2020-06-16 by mahtDFR