Ultimaker/Cura #7933: [REQUEST] Rename items in Object List

KenLucke, 2020-06-14:
Please add the ability to rename the names of the items in the Object List.  This would help to keep track of duplicated objects better, and also allow the support blockers and other "attached" items to be named so that they could be identified.

Also, it would be helpful if things such as the support blockers would be hierarchical (preferably collapsible) with their parent object, rather than separate in the list.

smartavionics, 2020-06-15:
If you install the Mesh Tools plugin from the Cura marketplace, it has the ability to rename items in the object list along with its other useful features.

KenLucke, 2020-06-15:
> If you install the Mesh Tools plugin from the Cura marketplace, it has the ability to rename items in the object list along with its other useful features.

Ah, OK, have that installed, did  not realize it did that.  That helps... it would be easier to just control-click on the name to change it, but I can't have everythinig :)

The hierarchical stuff would still be cool.

mahtDFR, 2020-06-16:
Nice idea. We discussed it this morning and decided that as the functionality is already available as a plugin we can't prioritise this right now, so it is deferred and open to pull requests.

fieldOfView, 2020-06-16:
The next release will already have some improvements to the objects list (see https://github.com/Ultimaker/Cura/pull/7483). I was planning a PR to add "the hierarchical stuff" for a later release.

konskarm, 2020-10-22:
There is already a ticket in our backlog for this exact issue (CURA-6798). I am closing this issue as a duplicate of https://github.com/Ultimaker/Cura/issues/7116.

closed 2020-10-22 by konskarm