Ultimaker/Cura #7937: [4.6.1] Duplicate dialogue box when using "File Export Selection..." on MacOS Catalina followed by freeze (unresponsive)

mifga, 2020-06-16:
**4.6.1**
(The version of the application this issue occurs with.)

**Mac OS Catalina 10.15.5**
This bug seems to be specific to Catalina.

**Ultimaker S5**
(Which printer was selected in Cura?)

**Reproduction steps**
Entered on behalf of customer Greg Zumwalt.

1. With Ultimaker Cura 4.6.0, I select an object on the build plate.
2. Then select “File, Export Selection…”, and a file save dialog box appears.  
3. When I select the “Save” button on the dialog, a dialog appears saying the file already exists and asks if I wish to replace it.  
4. When I select “Replace”, yet another dialog box appears (redundant?) asking if I wish to overwrite it.
5. And then I select yes, then it saves it. 
HOWEVER
1. With Ultimaker Cura 4.6.1, I select an object on the build plate.
2. Then select “File, Export Selection…”, and a file save dialog box appears.  
3. When I select the “Save” button on the dialog, a dialog appears saying the file already exists and asks if I wish to replace it.  
4. When I select “Replace”, yet another dialog box appears (redundant?) asking if I wish to overwrite it.
5. And when I select yes, then Ultimaker Cura 4.6.1 freezes requiring me to quit Ultimaker Cura 4.6.1 and restart.  This did not happen before the upgrade of the S5 printers to firmware version 5.6.8

(not you might not get a redundant popup, but might get a freeze)

**Screenshot(s)**
Looks like Cura frozen.

**Actual results**
For some reason, Ultimaker Cura asks twice whether or not you wish to overwrite a file; once on the file save pop up (which is normal for MacOS), and a second annoying time after the file save pop up (which is not normal for MacOS).  With Ultimaker Cura running under MacOS Catalina, 90% of the time the second pop up does not appear, which is the indication that Ultimaker Cura has frozen and must be exited and restarted.

So first, we need to get the freeze bug fixed.  And second, it would be nice not to be asked twice if I wish to overwrite a file.

**Expected results**
That redundant dialog box seems like a bad sign, and freezing instead of saving the file is also frustrating.

**Project file**
Instead of project files -- because this happens even with brand new generated files, so doesn't seem to be project file specific -- Greg would like to share short video snippets:"

[Ultimaker Cura Freeze.zip](https://github.com/Ultimaker/Cura/files/4784080/Ultimaker.Cura.Freeze.zip)

Here’s a low resolution video of the bug.  As can be seen, I select “File, Export Selection…” on the menu bar, after which the “Save to File” popup menu appears.  I then select “STL File (Binary) (*.stl) from the file menu, then click “Save”.  A pop down menu appears on the save to file popup stating "“Gear, Spur (1m, 20t).stl” file already exists.  Do you want to replace it?” to which I click on “Replace”.  Normally, at that point, if all is well the second annoying file overwrite pop up would appear but as can be seen in the video it does not which indicates that Ultimaker Cura has frozen.  While watching the video, note I’m moving the cursor around the Ultimaker Cura window clicking on the various items (e.g. translate, rotate, deselect, etc.) without success, thus the Ultimaker Cura application window has froze.

Two problems indicated here.  First, the second annoying file overwrite prompt, and second (the most important), Ultimaker Cura freezes and must be exited in order to resume normal operation.

And second video:
[Ultimaker Cura Freeze 2.zip](https://github.com/Ultimaker/Cura/files/4784084/Ultimaker.Cura.Freeze.2.zip)

Here’s a second video showing a gear I designed on the Autodesk Fusion 360 X, Z axis in .STL format that loads into Ultimaker Cura on edge thus it must be rotated 90 degrees to lay flat for 3D printing.  Note once rotated, I can easily print the gear.  But when I attempt to save the gear in its new orientation (for publishing) note the dual overwrite questions (the second is annoying) but as the second overwrite question appears, it indicates that this time Ultimaker Cura did not freeze.  However, next note I load a second .STL into Ultimaker Cura and simply attempt to save it, where you’ll note the annoying second dialog does not appear which indicates that Ultimaker Cura has frozen and must be force quit then restarted again.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mifga, 2020-06-16:
Here are some additional insights into this issue from partner Jeremy Simon from 3DUniverse. He doesn't have Greg's files, but he was able to duplicate the error:

[JeremySimon_exampleLog_project_STL_for error.zip](https://github.com/Ultimaker/Cura/files/4787817/JeremySimon_exampleLog_project_STL_for.error.zip)

'I was able to duplicate the issue on my end (I'm also running Mac OSX Catalina). So my log file is attached. The end of the log file shows where the freeze occurred. I have confirmed that this issue occurs when you try to use File / Export and overwrite an STL file that is already present. The first "replace" confirmation appears normally, but then Cura hangs when the second redundant confirmation would normally appear.

'I have also confirmed that this issue does not occur when saving project files. I was able to use File / Save to overwrite an existing project file without any issues.

'Attached is the project file and STL file I used for testing this, but I don't think the project file will be of much use, since the issue only occurs when using File / Export or File / Export Selection.'

Ghostkeeper, 2020-06-17:
It's probably the same, or a very similar, issue as https://github.com/Ultimaker/Cura/issues/6643.

The Qt version we're using doesn't support Catalina since Catalina was released after the latest Qt 5.10 patch was released. We're looking into upgrading to a newer Qt version, but that means dropping support for older MacOS versions. (Developers, see issue CURA-7025.)

mahtDFR, 2020-06-22:
We found someone running Catalina, but unfortunately we still can't reproduce this issue.

jesimon, 2020-06-24:
I can confirm that I'm experiencing the same issue here. I'm running Mac OSX Catalina v10.15.5.

The issue only occurs when using File > Export or File > Export Selection and overwriting an existing STL file with the same name. Using File > Save does not cause the issue, even when overwriting a file of the same name.

mifga, 2020-08-25:
Greg has reported that this issue is still happening with 4.7.0 with his Catalina OS. To reproduce this error, do this:
1) Load an .STL into Ultimaker Cura 4.7.0.
2) Select the .STL by clicking on it.
3) Select “”Export Selection…” from the File menu.
4) Set the file format to “STL File (Binary) (*.stl).
5) Press “Save”.
6) Repeat steps 2) through 5) a few times until Ultimaker Cura 4.7.0 “freezes” whereupon which you will need to force quit the application.

aabmar, 2022-01-24:
Still present in 4.13.0. I'm using Monterey. The double replace will appear on all save operations, when you choose to overwrite a file. The first dialog is the OS native dialog, the second is a QT dialog.