Ultimaker/Cura #7938: Coasting does not apply for small extrusions, even though it is told explicitly to do so

thanhtacles, 2020-06-16:
I found that small extrusion lines were building up massive blobs and zits from nozzle pressure, and then I found out exactly why: the coasting setting was not behaving as expected. I even set the minimum threshold way down, and it refuses to behave. This fix can be pretty easily applied too. 

[image]

Like, you can clearly see that coasting is on and works, esp for outer wall. But outright ignores inner wall if it's small enough. Other larger inner-wall extrusions are unaffected.

rburema, 2020-06-16:
> Coasting replaces **the last part** of an extrusion path with a travel path. The oozed material is used to print the last piece of the extrusion path in order to **reduce stringing**.

From the hover-over description of 'Coasting', emphasis mine.

Coasting is not really the way to reduce overextrusion. You could try to reduce the flow or printing temperature instead.

thanhtacles, 2020-06-17:
Ok so by sheer luck I managed to figure out how to get coasting to apply. Will follow up with how once I figure it out 
[image]

Ghostkeeper, 2020-06-17:
If you fill in the template (in particular the printer you're using and the project file) we may be able to help you see why coasting is not applied here and to find any bugs you're running into.

closed 2020-07-01 by no-response[bot]