Ultimaker/Cura #7939: Gyroid direction ignored

BagelOrb, 2020-06-16:
**Application version**
4.6.1

**Printer**
UM3

**Reproduction steps**
1. slice with gyroid support infill 
2. observe direction of pattern on any particular layer
3. change the Support Infill Line Directions setting
4. observe directions of pattern on the same layer

**Screenshot(s)**
[image]
[image]

**Actual results**
Direction of pattern remains unchanged

**Expected results**
Pattern is rotated according to the change in the directions setting.

**Additional information**
Implementation: it's simply a matter of calling `area_polygons.apply(rotation_matrix.inverse())` before and calling `infill_line_segments.apply(rotation_matrix)` afterwards.

smartavionics, 2020-06-16:
Curious, it works for me...

[image]
[image]

smartavionics, 2020-06-16:
And at 90 deg too...

[image]

I think perhaps this is because my last round of updates to the gyroid and other TPMS infills didn't get added to UM Cura. Can't remember why not, I probably just lost patience and cancelled the PR.

BagelOrb, 2020-06-16:
What version are you on?

smartavionics, 2020-06-16:
This is with my Cura builds, not UM's releases.

mahtDFR, 2020-06-16:
If you're able to direct me to a PR, I can give it a nudge. I'll put this up for discussion anyway.

smartavionics, 2020-06-16:
It's too late, I've had enough, I'm now only contributing critical bug fixes to UM.

mahtDFR, 2020-06-16:
ok.

mahtDFR, 2020-06-16:
I reproduced this in 4.6.1

Ghostkeeper, 2020-06-17:
That'd be this PR: https://github.com/Ultimaker/CuraEngine/pull/1173
It didn't get merged (yet) since it's basically 4 new features + a refactor wrapped in one huge changeset with 1800 lines of code. I guess that's where the [pull request etiquette](https://gist.github.com/mikepea/863f63d6e37281e329f8) comes in. [Basically](https://readwrite.com/2014/07/02/github-pull-request-etiquette/) [every](https://erikzaadi.com/2019/09/29/pull-request-etiquette---a-set-of-simple-rules-for-your-code-review/) [etiquette](https://essenceofcode.com/2016/04/24/pull-request-etiquette/) [online](https://www.atlassian.com/blog/git/written-unwritten-guide-pull-requests) [agrees](https://medium.com/better-programming/pull-request-etiquettes-for-reviewer-and-author-f4e80360f92c): it helps to keep changes small. That allows us to fit it in our planning or even review/test it in between our tasks without planning it in at all. Google outright rejects external pull requests of more than 500 lines. We have no rule like that, though it will take a long time before these things get the light of day.

I still hope that the changes can get merged in some day but without Smartavionics' help that gets more unlikely now. We're going to see how hard it is to fix this bug individually anyway.

mahtDFR, 2020-06-18:
This was investigated yesterday and it seems that not only gyroid infill support, but also gyroid infill itself is affected by this, which makes it a harder fix.

We decided in the standup that it is a problem because it's inconsistent with the other infill patterns, however it's not something we wish to give priority to right now so it's deferred. Happy to review pull requests in the meantime.

closed 2023-06-24 by github-actions[bot]