Ultimaker/Cura #7940: Notify when print is should be removed

BagelOrb, 2020-06-16:
**Is your feature request related to a problem? Please describe.**
Often my prints don't start for the sole reason that - even though I did remove my last print from the build plate - I did not confirm that to the printer. I can lose several hours before I notice that the print hasn't started, which means that my print will also be done several hours later than it could have.

**Describe the solution you'd like**
In case there is a job in the queue, and in case the last print is ready, but not yet removed from the build plate, I would like to get a notification in Cura that I should (confirm to) remove the print from the build plate.
(This means that the notification would pop up at the moment you send a new job to a queueless printer which has a finished print on the build plate.)

Maybe add a button [Take me to the queue]

**Describe alternatives you've considered**
- Show confirmation msg in Cura when the print does start instead of when it doesn't.
- Show whole queue automatically when sending new job to printer
- Show notification in Cura only when you send a job

**Affected users and/or printers**
Ultimaker 3, and Ultimaker S line printers

**Additional information**
I really like this new feature template!

mahtDFR, 2020-06-16:
Personally I like this idea but in the stand up this morning we decided we don't want to do it because a notification system already exists

BagelOrb, 2020-06-16:
What do you mean? Are you saying I am already notified of it in some way? How?

mahtDFR, 2020-06-16:
If you're using a network connected Ultimaker you should get a notification when a print is complete. Also if you use the app, there are notifications there too.

...Admittedly this is slightly nuanced from what you've suggested.

BagelOrb, 2020-06-16:
I've never seen that notification. Does it disappear 30 seconds after it appeared?

Whatever man, I'm used to issues being discarded during standup without too much thought.

I'm not gonna pretend I can change that any more.

closed 2020-06-16 by BagelOrb