Ultimaker/Cura #7941: Please add the option to change from Metric to SAE in Scale measurements

GoguyT3D, 2020-06-16:
I would like for there to be the option from possibly a dropdown menu or from within the settings for the measurement description of the size/scale of the object to be changed from metric(mm) to SAE(inches)
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)
I think that all users who live in the US and use measurements in inches will greatly benefit from this because the option will allow for more accurate scaling when changing or pre-printing an item. Considering they will be measuring in inches rather than having to convert to verify that the measurements are correct.
**Additional context**
(Add any other context or screenshots about the feature request here.)

rburema, 2020-06-17:
> Considering they will be measuring in inches rather than having to convert to verify that the measurements are correct when the % scale of an item has been changed from the original file.

As it stands now, they can use the Barbarian Plugin to convert.

Otherwise, it's been our position for a long time that two measuring systems within one package will cause more trouble than its worth. See for example [this thread](https://community.ultimaker.com/topic/28414-is-it-possible-to-reset-cura-from-mm-to-inches/) on the forums.

GoguyT3D, 2022-10-25:
I seemed to have found a workaround using Autodesk Fusion 360 with SAE and choosing to convert to MM during the import process, Thank you

closed 2022-10-25 by GoguyT3D