Ultimaker/Cura #7942: Prime Tower Size alarm not justified (orange highlighted)

GalaxAjh, 2020-06-16:
**Application version**
4.6.1

**Platform**
Mac OSX 10.13.6

**Printer**
Dagoma Disco Easy 200 (Dual Extruder)

**Reproduction steps**
I have two objects to print with two material. Prime tower is enabled. The size is 20 mm (default size). As you can see on the screen shot, a little space is reserved for extruder and there is no warning about the size of the tower.
On the second screen shot, I have set the size of the prime tower to 27 mm. As you can see, the size of the tower  is highlighted in orange. I suppose that it is a warning. And yet, it is just the good size for the tower (3rd screen shot)

**Screenshot(s)**
[image]
1°) no warning and the prime tower is to small
[image]
2°) warning  and the prime tower have the good size
[image]
3°) You can see at the center, there is juste a little space to be sure that the purge/prime is completed

**Actual results**
Described above

**Expected results**
A warning about the size of the tower depending of the prime tower volume, when the size of the tower is too small (and no warning when it is big enough)

**Project file**
I can share if needed

**Log file**
I can share if needed

**Additional information**
About flying help on Prime Tower Size, it says "affect Prime Tower Y Position".
I think, you can add "affect Prime Tower X Position"
and "affect the Prime Tower Volume"

** Questions relative to the Prime Tower**
The data sheet of my nozzle says Purge : 55 mm2. I Suppose, this a the value that I must set in Prime Tower Minimum Volume. So what is the "_Nozzle Switch Extra Prime Volume Amount_" I have set this value to 55 mm2 to, but if I put 0 for this value, I do not see any change on the simulation.
Sorry to annoy

V.

rburema, 2020-06-17:
Right now it'll always warn when larger than 20mm (no formula ... by default at least, machine definitions & the like can override this).

I'll ask around what the reason is, if any.

rburema, 2020-06-17:
We decided to fix it for now with a larger arbitrary number. See commit 7e18e796d.

rburema, 2020-06-17:
> I do not see any change on the simulation.

The produced gcode is definitely different though! Probably because priming doesn't really register quite the same way as a travel or normal extrude.

> Sorry to annoy

There's no shame in asking good faith questions! :-)

rburema, 2020-06-17:
> About flying help on Prime Tower Size, it says "affect Prime Tower Y Position". I think, you can add "affect Prime Tower X Position" and "affect the Prime Tower Volume"

Those are automatically generated: Try adjusting the size and see what formula's change: It's only the Y-pos that does. The x and _minimum_ volume are not affected.

closed 2020-06-17 by rburema

Ghostkeeper, 2020-06-18:
Link to correct commit: https://github.com/Ultimaker/Cura/commit/9204ed4a02d5c4c47d2eb7037827d825e28dd12d