Ultimaker/Cura #7943: Top/Bottom Thickness error thickness if options Use adaptive layers enabled

lk29, 2020-06-16:
cura  4.6.1 

Win 10/64

Use adaptive layers enabled

If the upper or lower shell coincides with the function of the adaptive layer height, the layer height in mm is not calculated correctly.

Probably a pre-calculated height calculation is used before the function is triggered.

mahtDFR, 2020-06-17:
A project file or screenshot would help here, could you provide these please?

(`File > Save...` to save a project file)

rburema, 2020-06-17:
Sorry, accidentally closed. Wrong issue number in commit!

lk29, 2020-06-17:
[image]
[image]
[image]

lk29, 2020-06-17:
[Cura_BUG_D_VP_Refined_Chamber_03.zip](https://github.com/Ultimaker/Cura/files/4791771/Cura_BUG_D_VP_Refined_Chamber_03.zip)

lk29, 2020-06-17:
Layers 90-93 thickness 0.1 4*0.1 = 0.4 
should be 0.8

lk29, 2020-06-17:
[image]
[image]

rburema, 2020-06-18:
Hi @lk29 

Thanks for the info!

So it turns out this has been a known issue for a while, and is in fact a duplicate of #3913 we have an internal ticket for it, CURA-5104. Other related tickets are #3209 #4396 #5131 and #6188 

This is basically the big issue preventing us from moving adaptive layer heights out of 'experimental', not only because it would require a refactor of the engine in some key parts (possibly the biggest issue), but also because it isn't clear then whether the number of layers or the thickness of the vertically specified setting should be leading (what if the user specifies number of layers vs. thickness of feature in mm. or vice versa, or both).

We hope to start working on this at some point though. It's a matter of prioritisation.

Since this is a duplicate however, and the original ticket is still open, please continue any discussion over there.

commit 9204ed4a (2020-06-17): Fixes #7943 / Replacing an arbitrary value with a better one.

Not the most elegant way, but it _is_ the ultimate answer after all.
closed 2020-06-18 by rburema