Ultimaker/Cura #7944: I don't think this a duplicate. On mac there is no option for the nozzle size on in the ender3 template. dropping files into variants does nothing because there is no entry.

bakaufman, 2020-06-17:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

bakaufman, 2020-06-17:
https://github.com/Ultimaker/Cura/issues/7571#issue-605089533

Ghostkeeper, 2020-06-19:
This was fixed in Cura 4.6.1. As you reported that you have this issue but didn't fill in the template, I'm assuming that you are still using Cura 4.6.0. Please upgrade.

closed 2020-06-19 by Ghostkeeper