Ultimaker/Cura #7945: Multiply grouped objects never ends

danielo515, 2020-06-18:
**Application version**
4.2.1

**Platform**
MACos

**Printer**
Witbox

**Reproduction steps**
1. Group several objects. In my case I grouped 3, 2 had per-object settings for a different infill of the first one
2. Select the grouped object and select multiply
3. The process never ended, and I left it one entire day

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
The multiplying and placen object modal and progress bar never fills and never ends, but the program still works.

**Expected results**
Well, a new group of objects with the same relative positions

**Project file**
Attached
[ultimate-spool-holder.3mf.zip](https://github.com/Ultimaker/Cura/files/4796855/ultimate-spool-holder.3mf.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I have the custom supports plugin installed

nallath, 2020-06-18:
I just tried reproducing this with your project file, but it works fine for me. Could you also add your log files? There might be something there that can give us some more infor what went wrong.

danielo515, 2020-06-22:
Sure thing. Could you please link me to instructions about how to get logs on OSx? The official docs seems to be only for windows.

nallath, 2020-06-22:
You can find that here:  https://github.com/ultimaker/cura/#logging-issues

closed 2020-07-06 by no-response[bot]