Ultimaker/Cura #7948: Adaptive Layers/Tree Support

TampaBilliam, 2020-06-18:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
CR-10S

**Reproduction steps**
Hello, thought I would report what seems to me to be a bug with Tree supports and or adaptive layers. When I have tree support turned on with adaptive layers the supports will leave entire sections without support at all and will start new tree supports in mid air. If I turn adaptive layers off, the unsupported areas get support again....but there are still some tree supports starting up in this air halfway up the print. I'm attaching two screen shots of one of the unsupported areas and another of the tree supports starting in the air halfway up the print. I didn't notice it in preview and was 12 hours or so into a print when it was printing in thin air. Then I saw it in the preview. It is repeatable. 

**Screenshot(s)**

[image]
[image]
[image]

**Actual results**
Spaghetti model due to mid air printing with no supports.  

**Expected results**
supports should support the entire project where warranted and base on the bed or the model, not air.  

**Project file**

Cura's bug section won't accept a .3mf file from Cura?   That just seems weird to me.  I'm happy to email you the stl, or the 3mf....not sure what else to do here.   I don't see a file->save option.  

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-06-24:
> Cura's bug section won't accept a .3mf file from Cura? That just seems weird to me. I'm happy to email you the stl, or the 3mf....not sure what else to do here. I don't see a file->save option.

We're using github as our bug-reporting platform. The restrictions are theirs, not ours. _However_ it does seem to accept these files if you zip them (or even rename them to zip).

In fact, I suggest that you do just that, as issues like these are otherwise (without a reproducible file) _very_ hard to solve.

Ghostkeeper, 2020-06-26:
The "starting halfway in mid-air" part is due to a high Support X/Y Distance I think.

The unsupported ridge shouldn't really be dependent on layer height. A project file would definitely help to figure out what's going on there.

TampaBilliam, 2020-07-01:
Hey guys, sorry it took a bit...was traveling and had to get back to computer to zip the file.   With tree support on the rounded section of this file is not supported over a large gap.   Turn on adaptive as well and there are several sections down the sides as well.   Turn both of them off and regular support on and everything is supported as you would expect.    Still love your slicer, but thought I would point it out since all input just makes it better.  

[CuraTroubleShooting.zip](https://github.com/Ultimaker/Cura/files/4857676/CuraTroubleShooting.zip)

Ghostkeeper, 2020-07-06:
I'm not seeing that same unsupported ridge when I try it with your project file :confused:

[image]

It's supported fine, albeit with a bit too much support interface if you ask me. A known [limitation](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/adaptive_layer_height_enabled.md) of Adaptive Layers is that it doesn't have correct thickness everywhere.

closed 2020-09-09 by no-response[bot]