Ultimaker/Cura #7949: [4.6] problem in generating walls

mehran-Bgh, 2020-06-19:
**4.6**

**Windows 10**

**Ender 3**

I have designed a heart shape mold with Autodesk Inventor. after I generate the G-code and print it I realized some surface problems on the model. when I rechecked the preview of the Cura I realized the walls didn't generate in some simple parallel lines. 

**Screenshot(s)**
[image]
[image]
[image]

Generating this kind of wall increased the build time and ruin the quality of the surface. 
**Project file**
[CE3_Under side mold.zip](https://github.com/Ultimaker/Cura/files/4802634/CE3_Under.side.mold.zip)

smartavionics, 2020-06-19:
Unfortunately, the model has problems. Install the Mesh Tools plugin from the marketplace to get these warnings...

[image]

The model appears to have thickness to the walls but also a solid top and bottom, that's not going to work...

[image]

mehran-Bgh, 2020-06-19:
you are absolutely right. thanks my friend

closed 2020-06-19 by mehran-Bgh