Ultimaker/Cura #7950: Profile Settings are not  fully updated in "Preferences" on change

NilsRo, 2020-06-19:
**Application version**
4.6.1

**Platform**
Windows 10 X64

**Printer**
Anycubic I3 Mega s

**Reproduction steps**
1. Open Preferences
2. Show Profiles
3. Select another profile with diffrent settings

**Screenshot(s)**
[image]

**Actual results**
The settings shown are changed to the selected profile partly. E.g. the layer thickness but some other settings not like the temperatur. 

**Expected results**
Settings of the profile selected should be shown. This make it easier to compare profiles.

**Project file**
-

**Log file**
-

**Additional information**
-

nallath, 2020-06-19:
What do the settings in the extruder tab 1 say (in the preferences window)

Ghostkeeper, 2020-06-26:
The extruder tab is currently empty due to a different bug. I can already tell you that.

Ghostkeeper, 2020-06-26:
Looks like this is a duplicate of https://github.com/Ultimaker/Cura/issues/7626.

We haven't been able to reproduce the issue so far. Or rather, we could reproduce it by loading in a modified configuration folder, but that modification shouldn't be possible through the GUI. We don't know yet how to reproduce the problem from scratch.

closed 2020-06-26 by Ghostkeeper