Ultimaker/Cura #7954: [BUG] double click on project doesn't open it

GMagician, 2020-06-19:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
Custom

**Reproduction steps**
Double click on file on win explorer

**Actual results**
Cura opens, it shows for a short time the "open project" windows then it closes such windows without user action then opens the program window without any project loaded. Selecting open file now it opens with no issues

**Expected results**
It open the project

**Project file**
[Fan silencer.zip](https://github.com/Ultimaker/Cura/files/4805482/Fan.silencer.zip)

nallath, 2020-06-19:
Could you also provide the logs?

GMagician, 2020-06-19:
[stdout.zip](https://github.com/Ultimaker/Cura/files/4805919/stdout.zip)

GMagician, 2020-06-24:
Are the logs correct?

Ghostkeeper, 2020-06-29:
Yeah. Well, it doesn't mention anything about opening a file. This log file looks the same as if someone opened Cura without any provided file to open immediately.

I'm unable to reproduce this myself on Linux. If I open a project file with Cura, after booting up Cura it'll pop up with a dialogue for loading the project file.

Ghostkeeper, 2020-06-29:
Hmm, although it may be having trouble with the space in your file name.

Ghostkeeper, 2020-06-29:
Do you also get an error that says it can't open files of type "/path/to/Fan"?

GMagician, 2020-06-29:
You are right!!!!! I tryed to load the same file, with and without space on name and the one with space does exactly the issue I report here

GMagician, 2020-06-29:
> Do you also get an error that says it can't open files of type "/path/to/Fan"?

I don't recall such error. Where do I have to see it?

GMagician, 2020-06-29:
Here are log file loading "cable clip_horizontal.3mf
[Log.zip](https://github.com/Ultimaker/Cura/files/4846093/Log.zip)
"

GMagician, 2020-06-29:
mmm maybe not true....I was preparing log when name has no spaces but now it doesn't open it

GMagician, 2020-06-29:
@Ghostkeeper I did some more tests and thing are really weird.
1) Computer just switched on
2) I open "cable_clip_horizontal.3mf": it opens
3) I rename to "cable clip_horizontal.3mf": it doesn't open
4) Rename back to "cable_clip_horizontal.3mf": it doesn't open
5) Turn off computer
6) Turn on computer
7) I open "cable_clip_horizontal.3mf": it opens

from logs (err and out) I can't see anywhere the word "cable" so it seems it's not reported there neighter when it opens nor when it doesn't

Ghostkeeper, 2020-06-29:
> I don't recall such error. Where do I have to see it?

This is what I see:
[image]

GMagician, 2020-06-29:
No, nothing like that

GMagician, 2020-07-17:
It seems solved in 4.6.2

closed 2020-07-17 by GMagician