Ultimaker/Cura #7955: Usb connection error after changing to marlin

g2721, 2020-06-20:
**Application version**
4.6.1

**Platform**
MacOS 10.15.5

**Printer**
Anycubic i3 mega s (running marlin 2)

**Reproduction steps**
After upgrading to marlin Cura can't properly connect to the printer. The printer shows up in the Monitor panel, but can't be controlled neither can I upgrade the Firmware ("Firmware update failed due to communication error")

**Actual results**
[image]

**Expected results**
It should connect with out a problem. Repetier Host has no problem connecting to the printer and sending gcode.

**Additional information**
While the firmware update didn't change the baud rate (250000),  the autodetect feature could be the problem.
All other applications that could have used the USB port (like Repetier) were closed.
I got it working once after I updated (luckily, now I can at least print because motor rotation was flipped at first),but can't reproduce what I did different. 

Marc

fieldOfView, 2020-06-20:
We can't tell what is going on without logs.

g2721, 2020-06-20:
It seems like a clean install on an other computer helped, but it's quite frustrating.
Here is the log file:

[cura.log](https://github.com/Ultimaker/Cura/files/4807603/cura.log)

I would appreciate if you could fix this.
And for everyone else who stumbles across this problem: Try reinstalling Cura and when possible try using another computer.

nallath, 2020-06-22:
Uh, it looks like something was preventing Cura from opening any of the ports, since some other process was already using them.

Ghostkeeper, 2020-06-30:
Maybe there is something else communicating on the same port then? A different version of Cura running, or maybe the programmer you used to change the firmware?

breezy2u, 2020-07-05:
I'm having this exact same problem. I don't understand why this is so hard to fix. I am using the latest version of Cura and I have the firmware file I want to upgrade with, but it fails due to the connection error.

Ghostkeeper, 2020-07-07:
> I don't understand why this is so hard to fix.

Do you know a fix?
I can connect to my printer fine. Only for G2721, according his log, his operating system was saying that Cura is not allowed to talk to that port. We have not the slightest idea of why G2721's operating system is saying that. We're just guessing in the dark here, really.

g2721, 2020-07-08:
At the moment in don’t have access to my computer and the printer so it will take me another week until I can check what is blocking the usb port. Until then thank you for the support
Marc

g2721, 2020-07-08:
@breezy2u This is probably another problem issue. I guess in my case it’s a driver or access rights issue (because it worked well until I upgraded my printer to marlin; the first marlin install worked just fine)

morgano86, 2020-07-30:
I have a similar issue, I've always used Cura with my CR10s Pro on the Tiny machines firmware, During a 4.x upgrade Cura stopped detecting my printer. Every time I upgrade Cura I have to fix the issue by adding 250000 to the self._all_baud_rates array in the AutoDetectBaudJob.py file located in plugins\USBPrinting. The auto-detect by default does not seem to check 250000  and therefore whilst the printer shows up it's not able to communicate or print.

I'm not sure if this will fix your issue but it's worth a try. The file is protected so you'll need to grant full access using the windows security panel.

fieldOfView, 2020-07-30:
> I have to fix the issue by adding 250000 to the self._all_baud_rates array in the AutoDetectBaudJob.py

250000 is already the second one in the list:
https://github.com/Ultimaker/Cura/blob/4.6/plugins/USBPrinting/AutoDetectBaudJob.py#L21

morgano86, 2020-07-30:
Doh you are correct, looks like i got it working by removing all the other rates and leaving 250000:

`self._all_baud_rates = [250000]`

Ghostkeeper, 2020-08-03:
So trying first at 115200 has put the printer into a state where it won't respond to commands at different Baud rates?

morgano86, 2020-08-03:
I'm not sure why it works, but that sounds like a reasonable explanation.  I guess i could try putting the other rates back and putting 250000 first.

Ghostkeeper, 2020-08-04:
As a manual workaround that's doable, but wouldn't be ideal for most users :(

rrrlasse, 2020-09-29:
Maybe see if https://github.com/Ultimaker/Cura/pull/8463 works

morgano86, 2020-09-29:
> Maybe see if #8463 works

Seems like a reasonable fix.

g2721, 2020-10-25:
To summarize: I got it working flawlessly again by using a different computer with a clean cura install (both on win and macos)(reinstalling cura on my mac where the problem appeared didn’t help strangely enough) if you don’t have another computer lying around or it’s unpractical try playing around with different baudrates as described above. Also the last few updates might have fixed some stuff so updating cura, macOs and anycubic drivers could help.

closed 2020-10-25 by g2721