Ultimaker/Cura #7956: [master] random startup failures

smartavionics, 2020-06-20:
**Application version**
master branches as of 20/6/2020

**Platform**
Linux

**Printer**
NA

**Reproduction steps**
Start Cura (running from source, not appimage).

**Actual results**
Sometimes, not every time, Cura fails to start correctly and then it doesn't know about any printers or profiles and prompts the user to get started.

**Expected results**
Starts as normal, every time.

**Project file**
NA

**Log file**

2020-06-20 11:01:43,980 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin FileLogger
2020-06-20 11:01:43,985 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin UpdateChecker
2020-06-20 11:01:43,990 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2020-06-20 11:01:43,995 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SettingVisibilityProfileCreator
2020-06-20 11:01:43,997 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin Cura-PrinterSettingsPlugin
2020-06-20 11:01:44,002 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin Cura-LinearAdvanceSettingPlugin
2020-06-20 11:01:44,009 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin YappiProfiler
2020-06-20 11:01:44,683 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin Cura-MeshTools
2020-06-20 11:01:44,688 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin CustomJobPrefix
2020-06-20 11:01:44,691 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin FastView
2020-06-20 11:01:44,808 - DEBUG - [MainThread] RawMouse.RawMouse._start [157]: Imported <module 'hid' (/home/markb/.cache/Python-Eggs/hidapi-0.9.0-py3.5-linux-x86_64.egg-tmp/hid.cpython-35-x86_64-linux-gnu.so)>
2020-06-20 11:01:44,812 - DEBUG - [MainThread] RawMouse.RawMouse._start [175]: Found HID device with vendor_id = 256f, product_id = c652, interface_number = 0
2020-06-20 11:01:44,812 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [120]: Device 3Dconnexion / 3Dconnexion Universal Receiver, profile spacemouse
2020-06-20 11:01:44,812 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 0, scale = -0.100000, threshold = 0.010000, offset = 0.000000, target = movx
2020-06-20 11:01:44,813 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 1, scale = -100.000000, threshold = 1.000000, offset = 0.000000, target = zoom
2020-06-20 11:01:44,813 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 2, scale = -0.100000, threshold = 0.010000, offset = 0.000000, target = movy
2020-06-20 11:01:44,813 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 3, scale = -0.050000, threshold = 0.010000, offset = 0.000000, target = rotpitch
2020-06-20 11:01:44,813 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 4, scale = -0.050000, threshold = 0.010000, offset = 0.000000, target = rotroll
2020-06-20 11:01:44,814 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 5, scale = 0.100000, threshold = 0.010000, offset = 0.000000, target = rotyaw
2020-06-20 11:01:44,814 - DEBUG - [RawMouse] RawMouse.RawMouse._run_hid [216]: Trying to open /dev/hidraw0
2020-06-20 11:01:44,815 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin RawMouse
2020-06-20 11:01:44,815 - INFO - [RawMouse] RawMouse.RawMouse._run_hid [222]: Manufacturer: 3Dconnexion
2020-06-20 11:01:44,817 - INFO - [RawMouse] RawMouse.RawMouse._run_hid [223]: Product: 3Dconnexion Universal Receiver
2020-06-20 11:01:44,823 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SliceInfoPlugin
2020-06-20 11:01:44,827 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SolidView
2020-06-20 11:01:44,842 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin UFPWriter
2020-06-20 11:01:44,851 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin CuraDrive
2020-06-20 11:01:44,855 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin ImageReader
2020-06-20 11:01:44,859 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin PerObjectSettingsTool
2020-06-20 11:01:44,861 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [403]: Plugin [RemovableDriveOutputDevice] has been disabled. Skip loading it.
2020-06-20 11:01:44,864 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin ModelChecker
2020-06-20 11:01:44,869 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin PostProcessingPlugin
2020-06-20 11:01:44,873 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SentryLogger
2020-06-20 11:01:44,876 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin GCodeProfileReader
2020-06-20 11:01:44,879 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin AMFReader
2020-06-20 11:01:44,882 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade30to31
2020-06-20 11:01:44,884 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade22to24
2020-06-20 11:01:44,887 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade462to47
2020-06-20 11:01:44,889 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade46bto46
2020-06-20 11:01:44,892 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade35to40
2020-06-20 11:01:44,897 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade21to22
2020-06-20 11:01:44,900 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade27to30
2020-06-20 11:01:44,904 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade26to27
2020-06-20 11:01:44,907 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade40to41
2020-06-20 11:01:44,909 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade25to26
2020-06-20 11:01:44,914 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade44to45
2020-06-20 11:01:44,918 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade42to43
[... 397 lines cut ...]
2020-06-20 11:02:01,368 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/uranium/resources/shaders/selection.shader]...
2020-06-20 11:02:01,660 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/uranium/resources/shaders/select_face.shader]...
2020-06-20 11:02:01,719 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/uranium/resources/shaders/default.shader]...
2020-06-20 11:02:01,726 - DEBUG - [Thread-2] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for kossel_xl took 0.781130313873291 seconds.
2020-06-20 11:02:01,727 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/uranium/resources/shaders/composite.shader]...
2020-06-20 11:02:01,741 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/cura/resources/shaders/overhang.shader]...
2020-06-20 11:02:01,758 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/cura/resources/shaders/striped.shader]...
2020-06-20 11:02:01,775 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/cura/resources/shaders/transparent_object.shader]...
2020-06-20 11:02:01,786 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/cura/resources/shaders/striped.shader]...
2020-06-20 11:02:01,804 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/cura/resources/shaders/xray.shader]...
2020-06-20 11:02:01,810 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/cura/resources/shaders/xray_composite.shader]...
2020-06-20 11:02:01,832 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/opt/cura/share/uranium/resources/shaders/platform.shader]...
[... cut ...]
2020-06-20 11:02:11,509 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2020-06-20 11:02:11,510 - INFO - [MainThread] cura.CuraApplication.closeApplication [602]: Close application
2020-06-20 11:02:11,527 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura

**Additional information**

Looks like a race to me because sometimes it starts OK, sometimes it doesnt.

When it fails there are messages in the log saying "No active GlobalStack..."

nallath, 2020-06-20:
Setting version 10? Where did it get those profiles from?

smartavionics, 2020-06-20:
> Setting version 10? Where did it get those profiles from?

Oh, they were in some old .cfg files in share/cura/resources/intent. I have now removed them.

Were they a problem? I think they have been around for a long time but it's only recently that this startup issue has occurred. I'll let you know if the problem recurs or has gone away.

smartavionics, 2020-06-21:
I removed some of the out of date configuration files from my cura installation but still get the same random start failures..

2020-06-21 07:51:53,276 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin FileLogger
2020-06-21 07:51:53,281 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin UpdateChecker
2020-06-21 07:51:53,285 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [40]: Checking for new version of cura
2020-06-21 07:51:53,290 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SettingVisibilityProfileCreator
2020-06-21 07:51:53,293 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin Cura-PrinterSettingsPlugin
2020-06-21 07:51:53,298 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin Cura-LinearAdvanceSettingPlugin
2020-06-21 07:51:53,304 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin YappiProfiler
2020-06-21 07:51:53,927 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin Cura-MeshTools
2020-06-21 07:51:53,932 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin CustomJobPrefix
2020-06-21 07:51:53,935 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin FastView
2020-06-21 07:51:54,053 - DEBUG - [MainThread] RawMouse.RawMouse._start [157]: Imported <module 'hid' (/home/markb/.cache/Python-Eggs/hidapi-0.9.0-py3.5-linux-x86_64.egg-tmp/hid.cpython-35-x86_64-linux-gnu.so)>
2020-06-21 07:51:54,057 - DEBUG - [MainThread] RawMouse.RawMouse._start [175]: Found HID device with vendor_id = 256f, product_id = c652, interface_number = 0
2020-06-21 07:51:54,057 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [120]: Device 3Dconnexion / 3Dconnexion Universal Receiver, profile spacemouse
2020-06-21 07:51:54,057 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 0, scale = -0.100000, threshold = 0.010000, offset = 0.000000, target = movx
2020-06-21 07:51:54,057 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 1, scale = -100.000000, threshold = 1.000000, offset = 0.000000, target = zoom
2020-06-21 07:51:54,058 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 2, scale = -0.100000, threshold = 0.010000, offset = 0.000000, target = movy
2020-06-21 07:51:54,058 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 3, scale = -0.050000, threshold = 0.010000, offset = 0.000000, target = rotpitch
2020-06-21 07:51:54,058 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 4, scale = -0.050000, threshold = 0.010000, offset = 0.000000, target = rotroll
2020-06-21 07:51:54,058 - DEBUG - [MainThread] RawMouse.RawMouse._cacheProfileValues [139]: axis 5, scale = 0.100000, threshold = 0.010000, offset = 0.000000, target = rotyaw
2020-06-21 07:51:54,059 - DEBUG - [RawMouse] RawMouse.RawMouse._run_hid [216]: Trying to open /dev/hidraw0
2020-06-21 07:51:54,059 - INFO - [RawMouse] RawMouse.RawMouse._run_hid [222]: Manufacturer: 3Dconnexion
2020-06-21 07:51:54,060 - INFO - [RawMouse] RawMouse.RawMouse._run_hid [223]: Product: 3Dconnexion Universal Receiver
2020-06-21 07:51:54,061 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin RawMouse
2020-06-21 07:51:54,067 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SliceInfoPlugin
2020-06-21 07:51:54,072 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SolidView
2020-06-21 07:51:54,081 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin UFPWriter
2020-06-21 07:51:54,089 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin CuraDrive
2020-06-21 07:51:54,092 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin ImageReader
2020-06-21 07:51:54,096 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin PerObjectSettingsTool
2020-06-21 07:51:54,098 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [403]: Plugin [RemovableDriveOutputDevice] has been disabled. Skip loading it.
2020-06-21 07:51:54,101 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin ModelChecker
2020-06-21 07:51:54,104 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin PostProcessingPlugin
2020-06-21 07:51:54,106 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin SentryLogger
2020-06-21 07:51:54,109 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin GCodeProfileReader
2020-06-21 07:51:54,112 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin AMFReader
2020-06-21 07:51:54,115 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade30to31
2020-06-21 07:51:54,118 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade22to24
2020-06-21 07:51:54,121 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade462to47
2020-06-21 07:51:54,124 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade46bto46
2020-06-21 07:51:54,126 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade35to40
2020-06-21 07:51:54,131 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade21to22
2020-06-21 07:51:54,133 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [448]: Loaded plugin VersionUpgrade27to30
[... cut ...]
2020-06-21 07:52:09,791 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2020-06-21 07:52:09,815 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2020-06-21 07:52:09,815 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
2020-06-21 07:52:09,816 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [324]: Engine process is killed. Received return code -15

smartavionics, 2020-06-21:
I'm working on this. What I have found so far is that when it fails, it's because in MachineManager.setInitialActiveMachine() the call to findContainerStacksMetadata() is not returning any metadata for the active machine. When it does, the startup is OK.

smartavionics, 2020-06-22:
I have now found that when loading the metadata on cura startup, the current machine's metadata can either have container_type = ContainerStack in which case everything is hunkydory or it can have container_type = DefinitionContainer in which case the metadata later fails the search for a ContainerStack with that id. Now trying to find how that container can have different types.

nallath, 2020-06-22:
> Were they a problem? I think they have been around for a long time but it's only recently that this startup issue has occurred. I'll let you know if the problem recurs or has gone away.

It shouldn't be a problem, but it was the first thing I saw when I scanned the logs.

smartavionics, 2020-06-22:
Next discovery...

When it fails, I see this (my diagnostics)...

2020-06-22 09:19:46,197 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadMetadata [149]: SVO1 filename = /opt/cura/share/cura/resources/definitions/SV01.def.json, class = <class 'UM.Settings.DefinitionContainer.DefinitionContainer'>

But when it is OK, I see this...

2020-06-22 09:24:13,611 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadMetadata [149]: SVO1 filename = /home/markb/.local/share/cura/master/machine_instances/SV01.global.cfg, class = <class 'cura.Settings.GlobalStack.GlobalStack'>

As you can see, I have a definition of a printer called SV01 and also an instance of that printer called SV01 and, it's getting confused because the metadata's have the same id?

smartavionics, 2020-06-22:
Ah, fuck, I know what's changed recently, that SV01.def.json has appeared in the master branch, that's why this breakage has just started happening.

nallath, 2020-06-22:
That would cause it yes. The ID's must be unique.

smartavionics, 2020-06-22:
Hell, I've spent 2 days pissing about with this...

nallath, 2020-06-22:
Yeah, sorry :( I think we have automatic tests to see if newly added profiles have unique ID's, but that doesn't take your local global stacks into account.

smartavionics, 2020-06-22:
So the IDs must be unique even though the containers have different mime types?

nallath, 2020-06-22:
Yep.

fieldOfView, 2020-06-22:
(mumbles something about https://github.com/Ultimaker/Cura/issues/1085)

smartavionics, 2020-06-22:
OK, so I need to change my local defs for my SV01 but I have a bunch of project files that use that machine. Can I just change the id in my SV01.global.cfg to something else without it breaking all my existing profiles, project files, etc.

nallath, 2020-06-22:
If I remember correctly, the project loader checks for uniqueness when loading a container stack (and if it's not, it ensures that the new name is unique)

smartavionics, 2020-06-22:
I renamed machine_instances/SV01.global.cfg to machine_instances/SV01_mb.global.cfg and changed the value of id to SV01_mb but kept the name as SV01 and existing profiles and project files seem to understand. Phew! Closing...

closed 2020-06-22 by smartavionics