Ultimaker/Cura #7957: Tree supports

ardlc-1212, 2020-06-20:
Why dont you add a zigzag pattern or any other pattern inside the tree supports so they wont break during fast printing

jellespijker, 2020-06-20:
If you have problems with breakage of the trees when you're printing at a high speed, you could also try to increase the `Support Wall Line Count` or set the `Avoids Supports When Traveling`  if they get bumped over by the print head.

ardlc-1212, 2020-06-20:
> If you have problems with breakage of the trees when you're printing at a high speed, you could also try to increase the `Support Wall Line Count` or set the `Avoids Supports When Traveling` if they get bumped over by the print head.

thanks for the suggestion but I'm only giving an idea to improve it I don't actually need it

Asterchades, 2020-06-21:
If you just want a generic infill type thing than it's already possible - just not enabled by default. When you enable Tree Supports, all of the standard Support settings get enabled at the same time. So all you need to do is go up to the Support section and change the Support Pattern and Support Density as you see fit.

NB My experience has this occasionally putting the support pattern through the model itself on some layers, particularly when trees merge, so you'll want to check the layer view to make sure everything's clean. I haven't opened a separate report on this as trees are still experimental, so I figure glitches like that are known (or at least suspected) but not necessarily easy (or worth it) to fix.

jellespijker, 2020-06-23:
Is @Asterchades suggestion the same as your proposal?

Below an image of tree support with 10% zig zag support
[image]

Ghostkeeper, 2020-06-30:
Indeed this already exists, but is not the default for most printers (it is the default for some printers).

The tree support intersecting with the model sometimes is a known bug. We're putting some extra effort in the Tree Support in the near future to iron out such bugs from tree support and get it more stable.

closed 2020-06-30 by Ghostkeeper