Ultimaker/Cura #7958: 4.6.1 Monitor Right Extruder Temp displays on Left Extruder Temp Field

NEW-PRNTR, 2020-06-21:
**Application version**
4.6.1

**Platform**
Windows10 Home

**Printer**
Creality CR-X

**Reproduction steps**
1 - Select Pre-heat for right extruder

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
See Temp for Right Extruder in the right field

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2020-06-24:
By 'left extruder temperature field', do you mean on the printer or in Cura?

NEW-PRNTR, 2020-07-14:
Apologies, email scrolled away from main screen and missed replying.

Cura.

mahtDFR, 2020-07-16:
Hi @NEW-PRNTR 
This issue was discussed in our standup this morning. Unfortunately we've decided to defer it since we don't have dual extrusion printers with a serial connection to test with. The Ultimaker product portfolio has moved away from printing via serial so, it's not really a development priority to support that right now.

We can review pull requests from the community if someone is able to submit **and test** a fix.

closed 2023-06-13 by github-actions[bot]