Ultimaker/Cura #7959: Cura v4.6.1 Failed to import profile error

thor171, 2020-06-21:
Hi,
I'm using Cura version 4.6.1 on my computer which is running Ubuntu 18.04.
I tried to import a profile but I always get this error message:

"Failed to import profile from FILEPATH : Could not find a quality type for the current configuration."

What does this mean? I never had problems importing profiles before.
I added the profile, just changed the file extension from curaprofile to zip to be able to attach.

[A.zip](https://github.com/Ultimaker/Cura/files/4809326/A.zip)

nallath, 2020-06-22:
Your custom profiles are based on a quality profile. The error, in this case, indicates that it can't find the profile that it needs.

Ghostkeeper, 2020-06-30:
Looks like this profile is intended for the Prusa i3, quality level "fast" (called "Draft" in the interface). However since the Prusa i3 has no printer-specific quality profiles it's available for any other printer that doesn't have printer-specific profiles as well.

I added a Prusa i3 and imported your profile. It worked fine for me. Which printer are you trying to import it for?

closed 2020-07-14 by no-response[bot]