Ultimaker/Cura #7960: cura 4.6.1 crashes when opening any file or importing profile Windows 10

ghclifton, 2020-06-21:
Since upgrading to Cura4.6.1 from 4.5 i can no longer open any files or import any profiles.
have uninstalled and reinstalled many times and going back to an older version of Cura does not fix the issue .  All other programs work fine on the laptop and can access files with no issue. I like cura and now have no slicing software till this issue is fixed. tried to capture a screen shot but cura closes to quickly as a side note if i open cura 4.6.1 and do nothing it will run all night , i can also use market place without issue.

Thanks
Gary

**Application version**
Cura 4.6.1
**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
trying to import profile or open stl
Window opens showing file list freezes and closes Cura
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Window dialog opens, freezes and cura program closes
**Expected results**
either stl opens or or profile is imported

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4809452/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/4809455/stderr.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4809456/DxDiag.txt)

**Additional information**
(Extra information relevant to the issue.)

ghclifton, 2020-06-23:
As an update to this issue I can drag an STL file to the work area but i cannot save it or open another stl with the folder icon in cura. still cannot import profiles. When you click on import, windows file screen comes up and freezes, then closes immediately.

rburema, 2020-06-24:
All I can find int he logs is this:

```
2020-06-21 04:36:40,730 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [286]: The definition cache for definition creality_base failed to pickle.
2020-06-21 04:36:41,089 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [179]: Extruder [creality_base_extruder_0 #2] has already been added to this stack [Creality Ender-3]
2020-06-21 04:36:41,105 - WARNING - [MainThread] cura.Settings.ExtruderManager.registerExtruder [156]: Extruder [creality_base_extruder_0 #2] has already been registered on machine [Creality Ender-3], not doing anything
```

Your comment about going back not working also suggests this to be a config/profile etc. issue. You could try backing those up (and) then starting from a clean folder in the appdata. (Also do this for the older versions, as otheriwse the upgrade will run again.)

Ghostkeeper, 2020-06-30:
Those warnings are fairly normal. I get them all the time. The "failed to pickle" warning is not so common (only happens when there's a hard drive failure or Cura crashed while writing that file) but the other two happen on every start-up for me.

The log file doesn't really give any information here, because it sounds like Cura freezes completely and is then killed by the operating system. Cura doesn't get the chance to write anything in the log about what's happening.
The freeze seems to happen when it's creating the main window. Somewhere in Qt, that is. But where in Qt is a mystery. Hopefully resetting the configuration folder helps.

Serge-Antoine, 2020-07-06:
I seem to have the exact same problem. I installed Cura for the first time a few days ago and noticed the only way I could load a model in was by dragging and dropping the file I want to slice. At first I figured something might've went wrong during the installation process so I uninstalled it, deleted all files liked to cura in Program files and did the same with appdata. Reinstalled everything and nothing has changed. The log files don't seem to show anything (as mentioned by others).
It also crashes if I want to export gcode or save the project. In other words, it seems like any time cura has to open a file explorer window it crashes. (I don't know if such information can help the Devs but I'm saying it anyway.)
Luckily for me, I can still slice the models and send them to the printer via network without any problem so this issue is more of an inconvenience than anything else.
FYI my colleague also installed cura on his computer and it's working as intended so I have no idea why it wouldn't work for me.

Ghostkeeper, 2020-07-08:
Do you have any explorer extensions installed that could be causing a problem? Like Google Drive, Dropbox, Adobe, etc?

This was reported [here](https://github.com/Ultimaker/Cura/issues/1537). Sadly Qt closed their bug saying that it's Windows' fault.

lillalysmasken3, 2020-07-09:
I've recently had this issue a lot. Just turned off gogle drive sync and One drive. Seemed to help this time. Will investigate further.

smartavionics, 2020-07-09:
I don't know if this is related or not but a user of my Cura builds reported that they were crashing when the file open dialog is opened and it turned out that this was due to a different Qt5Core.dll being used. Here's the link to the solution comment...

https://github.com/smartavionics/Cura/issues/66#issuecomment-655699145

ghclifton, 2020-07-10:
ok here is what i have tried, I Deleted cura leaving the profile intact, went through the registry and removed any thing with Cura in it , Ran CC cleaner on the registry, re-installed Cura 4.6.1 allowing it to re-use profiles. Ran Cura as Admin and it worked was able to import some old Cura profiles and open some STL files. i have tried running as Admin before it didn't work and compatibility mode. One Drive is disabled during this run and same for google drive, i only use them from time to time but are disable for now.   Sorry it took so long to respond and i am not sure exactly what step fixed it. thank you for all your help, i will continue to look at this and let you all know if i can repeat it.....

mahtDFR, 2020-07-10:
@ghclifton Good that it's working again. Will close this ticket but if you have a solution you can add it here.

closed 2020-07-10 by mahtDFR

ghclifton, 2020-07-11:
thanks for the help and ideas, it was appreciated

lillalysmasken3, 2020-07-14:
I've recently had this issue a lot. Just turned off gogle drive sync and One drive. Seemed to help this time.

Serge-Antoine, 2020-07-30:
I uninstalled Dell's backup & recovery software thingy and the problem is gone! I guess it was some weird behaviour with Qt5Core.dll or something. Gotta love it when bloatware messes up your stuff.