Ultimaker/Cura #7961: Resolution for "curved" combing moves

CCS86, 2020-06-21:
Running 4.6.1 on Win 10

Issue: when performing interpolated, curved combing / rapid moves, the printer makes a low frequency "grunting" vibration. I believe this is because with accel/jerk settings that work very well for printing speed moves, when they are run at 180 mm/s, the coarseness of the linear interpolation sets up a low frequency harmonic with the timing of jerk heavy junctions. 

This is just a theory, but I think that changing the resolution of these moves could help break up any resonances. I'm happy to do some testing, if there is some way to change the governingsetting.

Thanks.

smartavionics, 2020-06-22:
Have you tried changing the Maximum Travel Resolution setting?

CCS86, 2020-06-23:
Oooof, that sounds like the exact setting I was asking for!

I never thought to look for it in the "mesh fixes" category. I expected "travel" or "experimental".

Let me do a little testing and see if this helps me out. Thanks!

closed 2020-07-01 by Ghostkeeper