Ultimaker/Cura #7962: Keep paused with "Not Responding" on Windows 10

jin-chong, 2020-06-22:
**Application version**
4.6.1

**Platform**
Windows 10 pro, ver 2004

**Printer**
Ender 3

**Reproduction steps**
1. Run cura
2. change print settings from right side a pull down widnow => paused with "Not responding"
3. after a few actions including changing view angle.  then do above second step again => paused with "Not responding"

**Screenshot(s)**
[image]

**Actual results**
Cura does not response showing "(Not Responding)" on the title.

**Expected results**
As I remeber, it didn't get paused like this when I installed v4.6.1 first.
I've reinstalled, and tried ver 4.5.  But they had same issue.

**Project file**
any project file.

**Log file**
2020-06-22 11:58:59,156 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-06-22 11:58:59,548 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [139]: Need to check for errors again. Discard the current progress and reschedule a check.
2020-06-22 11:58:59,741 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-06-22 11:59:00,276 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 0.5s
2020-06-22 11:59:09,346 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-06-22 11:59:53,595 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-06-22 11:59:54,097 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [139]: Need to check for errors again. Discard the current progress and reschedule a check.
2020-06-22 11:59:54,314 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-06-22 11:59:54,816 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 0.5s

**Additional information**
I appreciate you guys!  Thank you!

mahtDFR, 2020-06-22:
Can you share a full log with me?

mahtDFR, 2020-06-23:
We've encountered this before - usually its caused by out of date QT. Are you able to try updating?

fieldOfView, 2020-06-23:
> its caused by out of date QT. Are you able to try updating?

Unless OP is running Cura from source, no they are not able to update because QT is included in Cura and no matter what OP updates, the version of QT that comes with Cura is used.

closed 2020-07-06 by no-response[bot]