Ultimaker/Cura #7963: Cura dosnt uses all my RAM (it gets slow to slice)

agunievass, 2020-06-22:
hi, first i need to say that my english isnt perfect; im from Argentina, i have more or less knowledge of 3D Printing; but a lot of knowledge 3D Designing
2 Problem
the First Is: I have a big STL file; its arround 300MB, yeah, its heavy...
When i try to open it with Ultimaker Cura, it doesnt show up, its stays loading... and never ends.
i let it load for like 3 hours and still going; the "line of charge" isnt move...

The other thing is that when i look at "Administrador De Tareas" (a window wich shows the programs that are runing and what their are consuming in terms of RAM or CPU, and other things) there it shows that Ultimaker Cura is using 400Mb of RAM. This is very Low, i mean; with others programs that i use to handle Big STL uses like 3.000MB of RAM... my PC can handle it
So... there is a way to increase the "Slicing Capability" of Cura to make it load the STL files like real quick? Its like the Ultimaker Cura is locked on a max of 700MB of RAM... Somewhere over there... But not the 3.000MB that im witing for xd
so, there is a way to make Cura RAM "Unlimit"?

**Application version**
(( 4.1.0 ))

**Platform**
Im using Windows 10, Desktop PC; i have 8GB of RAM (arround 7GB usable)

**Printer**
i have a Custom printer; my one is Hellbot Magna 1 (like Ender 3); uses Marlin Code (and Gcode)

**Reproduction steps**
1. I Throw the STL File on Cura
2. it doesnt Load

**Screenshot(s)**
i Dont have screenshots, on Cura it shows "Loading /name_of_file/"

mahtDFR, 2020-06-22:
Thanks for the report. Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

If you haven't already, please can you attach a [log file](https://github.com/Ultimaker/Cura#logging-issues).

agunievass, 2020-06-22:
https://drive.google.com/file/d/1MdPco_KKIwt8HivkuXsDn3l2RAGX81eh/view?usp=sharing
This is the Link of the STL File that i used; i let it for like 5 hs and Cura load it, (shows it in Yellow) but when i tried to slice it... HA, my pc just cand handle it, it starts but inmediately gets laggy (Lagged) and it slows all my pc; after all i need to Restart my pc (i unpluged it and plug it again)
 i used 0.06 mm Layer Height
3 Shells
PLA Filament (215 Cº Extrussor Temp.) (50 Cº Bed Temp.)

After all i think that my PC maybe has a virus or one program is making my pc limited on RAM... Im not sure

agunievass, 2020-06-22:
https://drive.google.com/file/d/1MdPco_KKIwt8HivkuXsDn3l2RAGX81eh/view?usp=sharing
This is the Link of the STL File that i used; i let it for like 5 hs and
Cura load it, (shows it in Yellow) but when i tried to slice it... HA, my
pc just can't handle it, starts but inmediately gets laggy (Lagged) and
slows all my pc; after all i need to Restart my pc (i unplugged it and plug
it again)
i used 0.06 mm Layer Height
3 Shells
PLA Filament (215 Cº Extrussor Temp.) (50 Cº Bed Temp.)

After all, I think that my PC maybe has a virus or one program is making my
PC limited on RAM... I'm not sure

When i restarted the PC again (after unplugged it and plugged it again)
Windows tells me that has an internal error; and won't open, Windows tell
me that i can "Restore" from an after backup; i did that; but it simply
uninstalled all the programs, like "Factory Sealed" Windows... but keeps
the PDF files and that things (PDF, Images, and more. It only uninstalled
the programs)
For this reason I'm not sure if this is a Virus of my PC or when I
unplugged the pc something failed and became a chain of errors. i'm
pretty sure that Cura is not the cause of the failure; but idk, check it
from the STL
Lesson learned: Make Backups haha
Cheers and Thanks for replying

mahtDFR, 2020-06-23:
Ok, I wasn't able to reproduce this either. Will close this off then. Good luck!

closed 2020-06-23 by mahtDFR