Ultimaker/Cura #7964: Gap being added to wall section

nguse, 2020-06-22:
**Application version**
4.6.1

**Platform**
Mac

**Printer**
Creality Ender 3

**Reproduction steps**
The wall size is 2.4mm and I'm using a 0.4mm tip with a wall line count of 3 (other than increasing the wall line count from 2 to 3, no changes have been made to the default configuration from the standard print quality 0.2 profile).

**Screenshot(s)**
With addition:
[image]

Without addition:
[image]

**Actual results**
The "with-addition" model adds a tiny gap in the middle of the wall attached to the addition, large enough that it becomes two separate walls after the print completes.

The  "without-addition" does not have the gap and prints as expected.

The gap does not really show up until it's printed, but I noticed on the first layer of the preview, the slicer is adding a weird infill to one section of the wall.

**Expected results**
Gap added inside of the wall shouldn't exist.

**Project file**
STL files:
[models.zip](https://github.com/Ultimaker/Cura/files/4814494/models.zip)

rburema, 2020-06-25:
> The wall size is 2.4mm and I'm using a 0.4mm tip

The size of the tip isn't always precisely the width that gets deposited, depending on a whole host of factors, most of which are outside of Cura's control. This is why it says 'reducing line width slightly could result in better prints' on the hover-over text of 'line width'. In fact, on our own Ultimaker profiles, line-width is set to 0.35mm (or even lower in some instances for inner walls), even though the tip is also 0.4mm (other profiles are contributed by the community, and we can't test them here, so we just have to trust that they are correct).

This is (probably) also why it doesn't show up in the preview, but it might show up in your print.

> but I noticed on the first layer of the preview, the slicer is adding a weird infill to one section of the wall.

We're working on this ('weird infill') part (as well as a whole host of related issues & improvements) in the engine. It's a rather large change, so it might be a while before it makes it into a release though.

nguse, 2020-06-25:
Interesting, so by setting that line width to less than the nozzle, it's forcing some additional overlap between the walls then if I understand properly?

Ghostkeeper, 2020-07-01:
You understood that properly. It tends to make the walls a bit stronger at the cost of having a more equalised flow. Line width isn't necessarily the same as the nozzle size. However your profile doesn't do this (it just has a line width of 0.4mm everywhere) so that is off topic.

I think what you really just want to do is either to set "Fill Small Gaps Between Walls" to "Nowhere" or to enable "Filter Out Tiny Gaps". You're explicitly creating a gap between the walls here so you probably don't want that gap to be filled.

closed 2020-07-01 by Ghostkeeper