Ultimaker/Cura #7965: Add "Multiply model" to Edit menu (or unify context and edit menu?)

dbr, 2020-06-23:
It took me a while to find how to quickly duplicate a model several times - eventually I realised it was in the right-click context menu on the object ("Multiply selected object"). I was initially looking in the "Edit" menu where there is a bunch of similar tools (arrange, delete) but no multiply

Not using this feature often, I then forgot where this was and it took me a while to relocate it a few months later - thus my suggestion: could this action be added to the Edit menu? Would be easier to locate I think, as Cura doesn't appear to make much use of context menus so was not the first place I looked

Possibly the same with anything else in the context menu not already present in the Edit menu (mainly just "Center selected model"?)

It seems like this was intended to be there to some degree, the relevant(?) action does have `@action:inmenu menubar:edit`

https://github.com/Ultimaker/Cura/blob/61def4ba255840cb32f17cc3d2297bcb074358d3/resources/qml/Actions.qml#L281-L287

There's also another "Multiply Model" action with inmenu but no menu name - however doing the obvious thing and adding `menubar:edit` doesn't seem to fix this, and I'm not familiar enough with QML to debug this quickly

https://github.com/Ultimaker/Cura/blob/61def4ba255840cb32f17cc3d2297bcb074358d3/resources/qml/Actions.qml#L344-L348

I'm happy to look into this further if this change sounds reasonable and the cause isn't immediately obvious to someone familiar with the menu code!

mahtDFR, 2020-06-23:
Thanks for the request. This should be easy to do. I've an internal ticket to our current sprint: CURA-7524

evtrados, 2020-06-24:
This feature has been implemented and will be added to the next Cura release.

closed 2020-06-24 by evtrados