Ultimaker/Cura #7966: Show setting changes

BagelOrb, 2020-06-23:
**Is your feature request related to a problem? Please describe.**
I want to know which settings I have changed, so I can identify why my print is the way it is.
I would like to see an overview of the changed settings.

**Describe the solution you'd like**
In the profile dropdown there should be an option "Show current setting changes", which then opens a window very similar to the Discard changes dialog:
[image]

**Describe alternatives you've considered**
Have the same option in the dropdown, but it changes the setting visibility to show exactly those settings which have changed.

**Affected users and/or printers**
all

**Additional context**
After a couple of prints I forget what settings I am working with.

Current workaround is to change material and back.

mahtDFR, 2020-06-24:
I've added this to the backlog, CURA-7528

mahtDFR, 2020-06-24:
(Thanks for the request 😉)

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

tyeth, 2021-08-28:
Install the Profile Analyser plugin, actually called Profil Analyser... I'm using the Support Guide plugin too, wonderful, mouse over guide for each setting.
https://marketplace.ultimaker.com/app/cura/plugins/5axes/ProfilAnalyser

closed 2023-07-13 by github-actions[bot]