Ultimaker/Cura #7968: Software corrupted during computer restart or shut down.

GPHV, 2020-06-23:
Hey I couldn't find this issue in the search, sorry if it exists somewhere else.

Ultimaker Cura 4.6.1

Windows 64 bit

Creality Cr-10 V2

*Reproduction Steps*
Leave Cura open and restart computer or Cura is open when computer gets shut down for any reason.

*Problems encountered*
Cura attempts to open but after initializing its window goes white. If you click on the window at any point it says (not responding) and you have to open the task manager to end Cura.

**Expected results**
The app should open

**Additional information**
The current fix is to uninstall Cura and remove any files associated with it. Then to reinstall Cura. If every associated file is not deleted in the uninstall the reinstall only works for about 10 minutes before the app freezes, crashes and the problems stated above occur when trying to relaunch the app.

fieldOfView, 2020-06-23:
Do you happen to have multiple displays?

GPHV, 2020-06-25:
> Do you happen to have multiple displays?

If by that you mean multiple monitors for my computer yes, the laptop screen itself and a slightly larger Insignia TV monitor.

Ghostkeeper, 2020-07-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/4427. As a workaround, try disconnecting your second monitor, starting Cura then, and plug it back in. Also in the preferences you can disable the "Restore window position on start" then.

closed 2020-07-01 by Ghostkeeper