Ultimaker/Cura #7970: Cura 4.6 not launching in macOS 11 Big Sur beta

TimoLemburg, 2020-06-24:
**Application version**
4.6

**Platform**
macOS 11 Beta

**Printer**
not relevant

**Reproduction steps**
Launch Cura
Reinstall Cura and launch Cura

**Screenshot(s)**
not relevant

**Actual results**
Cura in not launching and is just quitting. 

**Expected results**
Cura should launch and show a regular Cura window

**Project file**
not applicable

**Log file**
Log file is empty, App has not been launched to a state where it can write logs.

**Additional information**
Launching Cura does not work.
Right-clicking on the Cura app to open it and bypassing the file quarantine doesn't work either.
Removing the quarantine flag doesn't help either.
The macOS log system shows entries regarding Cura:
   com.apple.xpc.launchd[1] (application.nl.ultimaker.cura.24069125.24069132[10009]): Service exited with abnormal code: 1
   com.apple.xpc.launchd[1]: Coalition Cache Hit: app<application.nl.ultimaker.cura.24069125.24069132(501)> [8459]
   No architectures specified in launch archictectures [ NULL ]  for app=file:///Applications/Ultimaker%20Cura.app/ which likely is an error.
   Sandbox: logd_helper(119) deny(1) file-read-data /Applications/Ultimaker Cura.app/Contents/MacOS/libgcc_s.1.dylib
   [app<application.nl.ultimaker.cura.24069125.24069132(501)>:11616] This process will not be managed.
   handle LS launch error: {\n    Action = oapp;\n    AppMimimumSystemVersion = "10.10";\n    AppPath = "/Applications/Ultimaker Cura.app";\n    ErrorCode = 1;\n}

I guess it is not launching because the App is not notarized (according to Taccy). I may be wrong here of course.

davidkrammer, 2020-06-24:
i do have the same problem. Also macOS 11 Dev Beta. Neither older Cura or newest Cura work.

jeffreyspaceboy, 2020-06-24:
Same problem

rburema, 2020-06-25:
I don't think the newer Macs support the version of Qt we're using, which we are stuck on ironically, because of slightly older Macs.

This is going to require some work and digging in the DevOps part in any case.

IngmarStein, 2020-06-25:
Could you check the system log whether you see messages from amfid when launching Cura? I've seen this problem with other apps and it was related to code signing.

nallath, 2020-06-25:
If it's the code signing, that would be great, since that is already high up on our backlog. So we would really appreciate if if someone can confirm that for us!

davidkrammer, 2020-06-25:
Where can i view get the logs for that? Then i would post them here.

rburema, 2020-06-25:
@davidkrammer You can find the logfile next to the `.cfg` file described here: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

fieldOfView, 2020-06-25:
@rburema, that is not the logfile he is asking for.

I think the best place to look for reasons the app does not start up (to the point that is does not even start logging) is in Console.app.

TimoLemburg, 2020-06-25:
I am seeing the following amfid entries in the log when I launch Cura:
standard	17:46:55.226214+0200	mDNSResponder	[R4299] DNSServiceCreateDelegateConnection START PID[115](amfid)
standard	17:46:55.226355+0200	mDNSResponder	[R4300] DNSServiceGetAddrInfo(C000D000, 0, 0, <mask.hash: 'Y3yJzKKuQ+Iird+3mbIEOg=='>) START PID[115](amfid)
standard	17:46:55.266771+0200	mDNSResponder	[R4299] DNSServiceCreateConnection STOP PID[115](amfid)
standard	17:46:55.266807+0200	mDNSResponder	[R4300] DNSServiceGetAddrInfo(<mask.hash: 'qnpsC+Qk4h3lNRf9kHdGEQ=='>) STOP PID[115](amfid)
It doesn't seem to be the issue.
Additionally I looked for the App Notarization (which is missing in Cura and should be one of the new must haves), but I have other Apps that are not notarized which are also still running.

TimoLemburg, 2020-06-25:
Here is what I could fetch in the console when starting Cura. I have removed the noise from other processes.

[macos11-log-cura-launch.txt](https://github.com/Ultimaker/Cura/files/4832461/macos11-log-cura-launch.txt)

rosskinard, 2020-06-26:
> Could you check the system log whether you see messages from amfid when launching Cura? I've seen this problem with other apps and it was related to code signing.

```
default	15:40:57.289305-0500	mDNSResponder	[R161518] DNSServiceCreateDelegateConnection START PID[120](amfid)
default	15:40:57.289395-0500	mDNSResponder	[R161519] DNSServiceGetAddrInfo(C000D000, 0, 0, <mask.hash: '6Cur6O9CppaeO/S6CYxFdw=='>) START PID[120](amfid)
default	15:40:57.290179-0500	mDNSResponder	[R161518] DNSServiceCreateConnection STOP PID[120](amfid)
default	15:40:57.290204-0500	mDNSResponder	[R161519] DNSServiceGetAddrInfo(<mask.hash: 'nlcB66KUburjY4sSDH0j2Q=='>) STOP PID[120](amfid)

```

That's all that showed up regarding amfid when I tried to launch Cura.

Everything in the system log with "cura" in the search...

```
default	15:40:57.277131-0500	runningboardd	Executing launch request for app<application.nl.ultimaker.cura.61963919.61963926(501)> (LS launch nl.ultimaker.cura)
default	15:40:57.277178-0500	runningboardd	Creating and launching job for: app<application.nl.ultimaker.cura.61963919.61963926(501)>
default	15:40:57.336447-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] This process will not be managed.
default	15:40:57.336477-0500	runningboardd	Now tracking process: [app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024]
default	15:40:57.336528-0500	runningboardd	Using default underlying assertion for app: [app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024]
default	15:40:57.336559-0500	runningboardd	Calculated state for app<application.nl.ultimaker.cura.61963919.61963926(501)>: running-suspended (role: None)
default	15:40:57.336600-0500	runningboardd	Acquiring assertion targeting app<application.nl.ultimaker.cura.61963919.61963926(501)> from originator [app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] with description <RBSAssertionDescriptor| "RB Underlying Assertion" ID:161-161-42591 target:30024 attributes:[
	<RBSDomainAttribute| domain:"com.apple.underlying" name:"defaultUnderlyingAppAssertion" sourceEnvironment:"(null)">,
	<RBSAcquisitionCompletionAttribute| policy:AfterApplication>
	]>
default	15:40:57.336702-0500	runningboardd	Assertion 161-161-42591 (target:app<application.nl.ultimaker.cura.61963919.61963926(501)>) will be created as active
default	15:40:57.337111-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] reported to RB as running
default	15:40:57.337610-0500	runningboardd	Calculated state for app<application.nl.ultimaker.cura.61963919.61963926(501)>: running-suspended (role: None)
default	15:40:57.337869-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Ignoring jetsam update because this process is not memory-managed
default	15:40:57.338088-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Ignoring suspend because this process is not lifecycle managed
default	15:40:57.338332-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Set darwin role to: None
default	15:40:57.339669-0500	loginwindow	-[PersistentAppsSupport applicationReady:] | App: Ultimaker Cura, ready, updating active tracking timer
default	15:40:57.338601-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Ignoring GPU update because this process is not GPU managed
default	15:40:57.339744-0500	loginwindow	-[ApplicationManager checkInAppContext:eventData:] | ApplicationManager: Checked in app : Ultimaker Cura
default	15:40:57.339412-0500	runningboardd	Acquiring assertion targeting app<application.nl.ultimaker.cura.61963919.61963926(501)> from originator [daemon<com.apple.coreservices.launchservicesd>:93] with description <RBSAssertionDescriptor| "foregroundApp:30024" ID:161-93-42592 target:30024 attributes:[
	<RBSDomainAttribute| domain:"com.apple.launchservicesd" name:"RoleUserInteractiveNonFocal" sourceEnvironment:"(null)">
	]>
default	15:40:57.339596-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Skipping AppNap state - not lifecycle managed
default	15:40:57.340130-0500	runningboardd	Assertion 161-93-42592 (target:app<application.nl.ultimaker.cura.61963919.61963926(501)>) will be created as active
default	15:40:57.342725-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Ignoring jetsam update because this process is not memory-managed
default	15:40:57.342982-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Ignoring suspend because this process is not lifecycle managed
default	15:40:57.343221-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Set darwin role to: UserInteractiveNonFocal
default	15:40:57.343426-0500	runningboardd	[app<application.nl.ultimaker.cura.61963919.61963926(501)>:30024] Ignoring GPU update because this process is not GPU managed
default	15:40:57.342622-0500	runningboardd	Acquiring assertion targeting app<application.nl.ultimaker.cura.61963919.61963926(501)> from originator [daemon<com.apple.coreservices.launchservicesd>:93] with description <RBSAssertionDescriptor| "foregroundApp:30024" ID:161-93-42593 target:30024 attributes:[
	<RBSDomainAttribute| domain:"com.apple.launchservicesd" name:"RoleUserInteractiveNonFocal" sourceEnvironment:"(null)">
	]>
[... cut ...]
default	15:40:58.542186-0500	loginwindow	-[PersistentAppsSupport applicationQuit:] | for app:Ultimaker Cura, _appTrackingState = 2
default	15:40:58.542220-0500	loginwindow	-[PersistentAppsSupport applicationQuit:] | App: Ultimaker Cura, quit, updating active tracking timer

```
I checked the cura.log and it hasn't been modified since I installed Big Sur, so the app never makes it far enough to actually writer entries to the file. I renamed it to even see if a new one would be created and it wasn't.

Let me know if I can provide any more info....
Thanks
Ross

IngmarStein, 2020-06-26:
It may have been a red herring then. In the other app's case, I saw the following items in the log
```
standard	00:16:59.865739+0200	amfid	Failure getting cert chain.
standard	00:16:59.865783+0200	amfid	Requirements for restricted entitlements failed to validate, error -67688, requirements: '<private>', error: (null)
standard	00:16:59.865802+0200	amfid	Restricted entitlements not validated, bailing out. Error: (null)
```
before it ends in a `No architectures specified in launch archictectures` similar to Cura on Big Sur.

christopher-wong, 2020-06-27:
+1 with this issue. I've attached my relevant logs from `Console.log`

[cura_4.6_console.log](https://github.com/Ultimaker/Cura/files/4839960/cura_4.6_console.log)

michaelrcarroll, 2020-06-27:
I retrieved the actual Cura stderr.log from ~/Library/Logs/Cura (stdout.log is empty).

Excerpt below. It looks like there's a problem actually loading [`shapely`](https://github.com/Toblerity/Shapely), a dependency of Cura's. `shapely` uses `ctypes` to load native libraries it needs, and `ctypes` is struggling (seems similar to [this StackOverflow post](https://stackoverflow.com/questions/62587131/macos-big-sur-python-ctypes-find-library-does-not-find-libraries-ssl-corefou)). Not sure how to address this, but hope this provides a little more context on what seems to be the root cause here.

```
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 697, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 51, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/cura/Arranging/Arrange.py", line 7, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/UM/Math/Polygon.py", line 10, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/UM/Math/ShapelyUtil.py", line 6, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geometry/__init__.py", line 4, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geometry/base.py", line 17, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/coords.py", line 8, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geos.py", line 113, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geos.py", line 56, in load_dll
OSError: Could not find lib c or load any of its variants [].
```

[stderr.log](https://github.com/Ultimaker/Cura/files/4841090/stderr.log)

Guilouz, 2020-06-28:
Same issue here

psterba, 2020-06-29:
I have a problem.

elliottxharris, 2020-06-30:
Same thing here

hywel-evans, 2020-07-08:
Still an issue on Beta 2

jarinudom, 2020-07-08:
> Still an issue on Beta 2

Yeah this likely won't get fixed in an OS update, because Big Sur changes the way software is allowed to look for library paths.

edvavi, 2020-07-10:
> > Still an issue on Beta 2
> 
> Yeah this likely won't get fixed in an OS update, because Big Sur changes the way software is allowed to look for library paths.

When someone have something to test let me know... Thanks.

gunterthomas, 2020-07-12:
Anyone tried to fix this?
I am working on a fix, but I am not able to compile Cura on MacOS 11 Beta 2.

Always exiting with "ImportError: No module named 'PyQt5.QtWidgets'" - even if simple Testcode including QtWidgets is working.

It would be cool, if anyone is able to get this tool running - as it is a public beta since a few days I guess we will have much more people trying to use Cura on Big Sur.

smartavionics, 2020-07-12:
From what I have read about this problem, the fix needs to be in Python. Whether that fix gets backported to the the version of Python that is used by UM is another matter.

gunterthomas, 2020-07-12:
This one sounds pretty bad....
So no way to fix this? Well ... any source to check to get some updates on this problem? Or do we have to sit here and wait?

nallath, 2020-07-13:
This is why I am of the opinion that Apple has an undying hatred for software developers. I mean, seriously, why the hell would you suddenly change the way that library path resolving works?

I don't really know if there is something we can do about this right now.

PelleRavn, 2020-07-13:
> This is why I am of the opinion that Apple has an undying hatred for software developers. I mean, seriously, why the hell would you suddenly change the way that library path resolving works?
> 
> I don't really know if there is something we can do about this right now.

You do know that Big Sur is a major release of MacOS (and not just a 10.x release)? Last time that happened was in 2001. 
If you had to do a breaking change, what better time (seen from a software development perspective) to do it with a major release? I could see what you would mean if it was a minor release (like 10.16).

This is a preview, that is under a month old, of what's coming, and not a stable release that have been out for years. You can't really blame Apple for you being trigger-happy for installing experimental software on your production machine. 

Sorry for being so direct. But relax, things will get fixed at some point! - Or downgrade back to MacOS Catalina!

nallath, 2020-07-13:
It would be the case if this was the first time that Apple pulled something like this. Unfortunately, it's not. OSX is by far the hardest OS for us to support (and keep supporting). This is just another thing on the list that makes it so. 

I'm personally on Fedora, so this change doesn't affect me at all (apart from having to fix it some day ;))

Ghostkeeper, 2020-07-14:
We're going to (have to) fix this, but there's a lot of hoops to jump through. Not the least of them is that it's extremely hard to get a virtualised environment for MacOS. And there's the hoop that we have to change our existing public keys with Apple in order to use the new signing system, but this needs re-verification and that takes a while.

For the particular issue that @michaelrcarroll posted an error for, it seems that @guntherthomas found a solution for that here: https://github.com/Ultimaker/cura-build/issues/242
Hopefully we can just apply the same change to our build system and it'll be fixed then.

TimoLemburg, 2020-07-14:
Okay, maybe I didn't catch a thing, but why is it extremely hard to get macOS virtualised? I have been running ESXi, Parallels, VMWare Fusion and VMWare Fusion Pro as well as Virtual Box with macOS. The catch is of course that you need a Mac in the first place to run those platforms on them.
I am working in macOS Client Designer and we use VMWare Fusion Pro for all our Mac testing.

smartavionics, 2020-07-15:
I have built a MacOS release of my Cura that incorporates the tweak to geos.py that is meant to fix this crash. So if someone could install the 20200715 release on Bug Sir and report back, it would be useful. My releases can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file before using.

TimoLemburg, 2020-07-15:
I tried your release and it starts (after getting the quarantine bypassed), launches and is using one processor core at full load. No start-up windows shows up and Cura is not reacting anymore (according to Activity Monitor). Activity Monitor also show increasing use or virtual memory, I stopped if after some minutes and about 80GB virtual memory usage.

TimoLemburg, 2020-07-15:
[Analyse von „Ultimaker Cura“.txt](https://github.com/Ultimaker/Cura/files/4923874/Analyse.von.Ultimaker.Cura.txt)
Here is the Activity Monitor analysis

smartavionics, 2020-07-15:
Thanks for trying it. That monitor file doesn't mean much to me, perhaps someone else can glean something useful from it.

Could you please provide the cura.log file? There may be something useful in there and also if there is a stderr.log file, provide that. Thanks.

TimoLemburg, 2020-07-15:
In my User Library there are only two empty log files for stderr and stdout, nothing else. Sorry, forgot to mention that.

smartavionics, 2020-07-15:
OK.

rburema, 2020-07-15:
> The catch is of course that you need a Mac in the first place to run those platforms on them.

... and therein lies the problem. Since we want to have all our (build) servers remote & do not want to be tied to a specific physical hardware item, which can get damaged, lost, stolen, suddenly stop working for an unknown reason, etc.

gothickitty93, 2020-07-15:
When I tried the build @smartavionics provided, It didn't work for me either. Same symptoms as reported by @TimoLemburg, and still doesn't start far enough to create a log file.

Ghostkeeper, 2020-07-16:
> In my User Library there are only two empty log files for stderr and stdout, nothing else. Sorry, forgot to mention that.

So that means that some Python code does get executed. It's a matter of figuring out where it's freezing.

niklasarnitz, 2020-07-16:
If you need someone to test things on the latest Beta, hit me up.

niklasarnitz, 2020-07-16:
These are my log files I get when opening Cura (just for more context):

stderr.log
```
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 697, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 51, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/cura/Arranging/Arrange.py", line 7, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/UM/Math/Polygon.py", line 10, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/UM/Math/ShapelyUtil.py", line 6, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geometry/__init__.py", line 4, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geometry/base.py", line 17, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/coords.py", line 8, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geos.py", line 113, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geos.py", line 56, in load_dll
OSError: Could not find lib c or load any of its variants [].

Original exception was:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/4.6/build/inst/bin/cura_app.py", line 191, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 51, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/cura/Arranging/Arrange.py", line 7, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/UM/Math/Polygon.py", line 10, in <module>
  File "/Users/ultimaker/build/4.6/build/inst/lib/python3.5/site-packages/UM/Math/ShapelyUtil.py", line 6, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geometry/__init__.py", line 4, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geometry/base.py", line 17, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/coords.py", line 8, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geos.py", line 113, in <module>
  File "/Users/ultimaker/build/env/4.6/inst/lib/python3.5/site-packages/shapely/geos.py", line 56, in load_dll
OSError: Could not find lib c or load any of its variants [].
```

niklasarnitz, 2020-07-16:
It seems like, Apple is still mapping libraries to paths, but has changed the way they do this: https://stackoverflow.com/questions/62587131/macos-big-sur-python-ctypes-find-library-does-not-find-libraries-ssl-corefou

niklasarnitz, 2020-07-16:
Okay, this should be addressed in the next python release: https://bugs.python.org/issue41100

jarinudom, 2020-07-17:
Python 3.8.4 is on Homebrew now, and based on the changelog it looks like it should fix the issue. Is there a way to get Cura to use the Homebrew python instead of the one bundled in the .app?

abrmecom, 2020-07-17:
3.8.4 doesn't include the fix, but I managed to get Cura running with 3.8.4 and some changes to find_library and libArcus (sip is now PyQt5.sip) so that's something I suppose.

abrmecom, 2020-07-17:
If anyone is willing to try, you could delete (or safely move) /Applications/Ultimaker Cura.app/Contents/MacOS/lib/python3.5/shapely/geos.pyc and add the included geos.py file in the same directory. Let me know how you get on.

[geos.py.gz](https://github.com/Ultimaker/Cura/files/4935355/geos.py.gz)

rosskinard, 2020-07-17:
> If anyone is willing to try, you could delete (or safely move) /Applications/Ultimaker Cura.app/Contents/MacOS/lib/python3.5/shapely/geos.pyc and add the included geos.py file in the same directory. [...]

Worked for me on Beta 2...dropped that file in and it launched. I haven't put it through any paces but it does launch now.

gothickitty93, 2020-07-17:
Thanks @abrmecom. That fix seemed to work for me as well. I'll report back if anything breaks when I have more time to test.

xiujunma, 2020-07-17:
> If anyone is willing to try, you could delete (or safely move) /Applications/Ultimaker Cura.app/Contents/MacOS/lib/python3.5/shapely/geos.pyc and add the included geos.py file in the same directory. [...]

Thanks, that works for me (Big Sur beta 2).

TimoLemburg, 2020-07-17:
Same here with the standard Cura 4.6.1.

niklasarnitz, 2020-07-17:
I just created a PR over at shapely. Maybe they'll include the fix in their latest release.

abrmecom, 2020-07-17:
It’s not really a workable patch, more a quick hack to get cura running on the beta.  The updates to python will fix the issue when they arrive so i doubt Shapely will want to change anything, but UM may need to do something if they stick with older python builds.

abrmecom, 2020-07-17:
[image]

A screenshot of the 4.6.2 download with patch (left) vs locally compiled with Qt5 and Py 3.8.4 etc.  Surprised the download looks so blurry, but I guess there is more to do than a simple Big Sur workaround.  Also notice the title bar colour (I use dark mode).

edvavi, 2020-07-18:
it does work!!!

edvavi, 2020-07-18:
works like a charm to me. Thanks!!!!

Ghostkeeper, 2020-07-20:
> A screenshot of the 4.6.2 download with patch (left) vs locally compiled with Qt5 and Py 3.8.4 etc. Surprised the download looks so blurry, but I guess there is more to do than a simple Big Sur workaround. Also notice the title bar colour (I use dark mode).

That may be because of a different version of Qt? We're using 5.10.2 in our builds. It works with newer versions of Qt (tested up to 5.14) but things like pixel alignment are sometimes different, causing this sort of issue.

Ghostkeeper, 2020-07-20:
I'll see whether we can merge that patch in (and what even changed between them). If that's acceptable, I think we'd post a MacOS build here to have you guys test it, because we don't have a computer with Big Sur to test on at the moment.

nallath, 2020-07-20:
We could also consider forking Shapely and applying the changes there.

gunterthomas, 2020-07-22:
You can simply clone Shapely - but I would suggest to create a Pull Request (which is already done -- https://github.com/Toblerity/Shapely/pull/947) and fix the broken parts, so it appears in shapely within the next release.
I would like to help with things here, but I am not deep enough into this whole ecosystem in python as I am (to my shame) working on the php site.
Seems to me like the PR needs some rework to be done (integration-/buildpipelines fail).

nallath, 2020-07-23:
> But I would suggest to create a Pull Request (which is already done -- [Toblerity/Shapely#947](https://github.com/Toblerity/Shapely/pull/947)) and fix the broken parts, so it appears in shapely within the next release.

I agree it's a bit better, but it also means that we have to upgrade the version of shapely that we use. Since there are more changes since then, it would require a fair amount of extra testing on our side.

Ghostkeeper, 2020-07-23:
We'll leave the pull request to the already existing one. No sense in making a second one with the same contents. I agree that it's preferable as long as we can update safely and in the past we've created such [pull requests](https://github.com/jstasiak/python-zeroconf/pull/205) to fix issues we had with our dependencies.

For our build system we can use a fork from the correct tag, or apply a patch file like we do to Qt to repair their file dialogues.

drelich, 2020-08-08:
Just installed the Big Sur public beta 1 and had the same problem. Your geos.py fix worked like a charm. Thanks so much!

[22 more replies]