Ultimaker/Cura #7972: Add possibility to build gcode from the model or other gcode, starting at the chosen layer and point, for resuming stopped prints.

Tygrys-1, 2020-06-24:
**Is your feature request related to a problem? Please describe.**
Sometimes my print stops or the nozzle clogs and then the nozzle start printing with air. If the print is sticked to the bed and not destroyed, it can be resumed with this procedure:
1) Remove the reason of failure.
2) Home the printhead.
3) Measure the height of the print (with a sheet of paper and the nozzle, just like leveling the bed), maybe find exact point of stop.
4) Run Cura, load failed gcode file, find the last printed layer - or even the place, where failure occured by animating the printhead and comparing to the top-level layer of the print.
5) Manualy edit the gcode file, to remove any G29, and the already printed part, add command to move to point, where things broken, add extruder offset.
6) Save changes and load the file to the printer.
7) Start printing the file and hope for the best :-)
**Describe the solution you'd like**
It will be easier, if the gcode file could be generated from the exact point, where the printhead is on the preview model. It should consist alternate preamble (the "start" gcode, as we need to heat up the printhead, but do not want to G29 or print anything below the surface of the model), also it should add save way to move the head to the desired starting point.

**Describe alternatives you've considered**
Maybe just generate the file from the point as above, any additional code could be added from the original file "starting" part, removing the G29's and skirts, brims and so.

**Affected users and/or printers**
Anyone, who have to trash the 80% made 24h print and was not gcode aware person.

**Additional context**
Well, it happens. We should save the Earth from the amount of trashed plastic.

Ghostkeeper, 2020-07-01:
Not saying I haven't tried this before, but resuming a print like that is kind of dangerous. If your measurement is off by a millimetre it can damage your gantry system.

I'm also slightly worried of people accidentally leaving their nozzle somewhere halfway on the print and then only getting half a print. Maybe this feature would create more trash than what it saves for the 1% who is expert enough to make it properly resume a print.

I'll propose it to the developer team to see what they think.

mahtDFR, 2020-07-02:
This has been discussed internally and we've decided we're not going to do it.

closed 2020-07-02 by mahtDFR