Ultimaker/Cura #7973: Cura 4.6.1 not showing Preview on Intel HD 520 (embedded)

spanzetta, 2020-06-24:
**Application version**
4.6.1

**Platform**
Samsung Laptop  NP900X3H-S03TR 
Intel Core i7-6500@2.5Ghz
Windows 10 Enterprise 1803 
Intel HD520 - Driver ver 25.20.100.6472 (OpenGL 4.5)

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Load a model
2. Click on "Slice"
3. Go to Preview Tab
4. You see nothing .. with Layer View.. if you select XRay view you see the model

**Screenshot(s)**
https://gofile.io/d/GYNnqk

**Actual results**
Preview is completely empty

**Expected results**
see the Preview 

**Project file**
with any project file

**Log file**
https://gofile.io/d/wPazZF

**Additional information**
nothing

spanzetta, 2020-06-25:
Update: if I enable "Force layer view compatibility mode (restart required)" the preview works..
At this point it can be considered closed (there is already a solution)

rburema, 2020-06-25:
It _should_ work with that version of OpenGL though.

I can see that the newer (non-'compatability') shaders _do_ fail, but the Intel driver fails to give the log, leaving us little chance to actually debug:

```
2020-06-24 16:55:08,049 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setVertexShader [121]: Vertex shader failed to compile: 
2020-06-24 16:55:08,055 - ERROR - [MainThread] UM.View.GL.ShaderProgram.setFragmentShader [132]: Fragment shader failed to compile: 
2020-06-24 16:55:08,061 - ERROR - [MainThread] UM.View.GL.ShaderProgram.build [148]: Shader failed to link: 
```

Let's just say that I wouldn't be surprised & this wouldn't be the first time the problem turns out to be 'Intel graphics drivers', to put it mildly :-/

spanzetta, 2020-06-25:
thanks for the feedback

in the meantime I marked it as solved because I realized that with the
"Compatibility setting" enabled it solved the problem

thanks again

regards