Ultimaker/Cura #7974: [4.5.0] Print head went through print in the end

4sfaloth, 2020-06-24:
**Application version**
4.5.0

**Platform**
Windows 10

**Printer**
Custom printer (Robo 3D R1+)

I've been using Cura to generate .gcode files to feed my printer for over a year, without any slicer issues I can't recall. But this week something weird happened: I was making a print with two objects, using the "**_One at a time_**" option. The printer started by printing the taller object (#1 on the image attached), and then the other. Both objects printed without any issue, but in the end, when the printer did the usual movement of 'return to base',  it just raised z by a little and then went back and to the left (REST on the image) as usual. Problem was the first print was in the way, so basically the print head collided with the print and broke it.

Is this a know issue? is it due to some mis-configuration on my part?
 I believeI have my print head dimensions introduced correctly

Please let me know if the .gcode or the cura log files are required to diagnose this.

Thanks in advance

[image]

FPrefect20, 2020-06-24:
Although i am not a CURA developer, but...

where does the 'return to base' command come from?
Can you please check your "end gcode" settings. (@printer settings)
This code in that section is not calculated by CURA but just added at the end.

4sfaloth, 2020-06-24:
> Although i am not a CURA developer, but...
> 
> where does the 'return to base' command come from?
> Can you please check your "end gcode" settings. (@printer settings)
> This code in that section is not calculated by CURA but just added at the end.

Hi @FPrefect20 , thanks for stepping in

I took a look at my end gcode section and in fact I have this in there
```
M104 S0 ; turn off temperature
**G1 X10 Y200**
M84     ; disable motors
```
so I guess that explains. I did not realize it was a direct instruction to go for a fixed position.

Is there any way to make it move taking the prints into account? What's the best practice here? I suppose leaving it in place oozing over the print is not the best idea?

FPrefect20, 2020-06-24:
What about switching the locations of your 2 objects and leave the end-gcode as it is?

The other option is to add an alternative end-position at your end-gcode section like this:
; G1 X10 Y200
**G1 X10 Y10**

You can leave the old code line with an ";" for easily switching the end position.

4sfaloth, 2020-06-24:
> What about switching the locations of your 2 objects and leave the end-gcode as it is?
> 
> The other option is to add an alternative end-position at your end-gcode section like this:
> ; G1 X10 Y200
> **G1 X10 Y10**
> 
> You can leave the old code line with an ";" for easily switching the end position.

Thing is I'm not really concerned with this specific print, but rather with any future prints I make with several objects. 
Do II really need to adjust this manually for every print? That seems a bit cumbersome. Perhaps there's any easy to at least pick the order at which it will print the objects, or something like that?

gailulun, 2020-06-25:
I think in this case, it is only necessary to raise the nozzle to the highest position of z-axis to start the nozzle from the model and avoid the nozzle colliding with the model, so you only need to take `G1 F1500 E-3; G0 F1500 Zxxx` as the last movement of the nozzle. Where "XXX" is the maximum value that your printer can raise the nozzle to, at this time, the final position of X and Y is uncertain.

gailulun, 2020-06-25:
or
```
G1 F1500 E-3
G0 F1500 Zxxx
G0 X0 Y0
```

Ghostkeeper, 2020-07-01:
Yeah I've always just taken the approach to move to Z200 or something, whatever the maximum Z is of the printer. It's best to move to Z first, not at the same time as X/Y, or you could still hit another object if it's taller than the last-printed object.

closed 2020-07-01 by Ghostkeeper