Ultimaker/Cura #7975: Ironing for User Selected Layers

exryan, 2020-06-24:
I would like to add a idea for allowing IRONING to important layers as selected by the user.  Say that surface finish is important on only a few layers, being able to specifically select these layers for ironing would be great.

Ghostkeeper, 2020-07-01:
Changing settings per-layer is something we've long held as a future option. It's really involved to implement though, both in the front-end and the back-end, and it's also something that is really quite expert and quite limited at the same time. I'd prefer to implement something like a [paint brush](https://github.com/Ultimaker/Cura/tree/feature_custom_support/plugins/CustomSupport) instead in the end. It's easier to understand for the user and more powerful too.

Currently you can select where you want ironing by adding a model that covers the layers that you want ironed (completely), setting that to change settings of overlap in the per-object settings tool, and enabling ironing for that object.

exryan, 2020-07-01:
Ahh! so just make a extra cylinder (or whatever shape) on the build plate that is as tall as the feature height and then the slicer will program a ironing task for the whole build plate at that height, then continue on to finish the part?  

Pretty good work around, thankyou for that advice!

Ghostkeeper, 2020-07-06:
Yeah, as long as it covers the entire layer you'll suffer no negative consequences. If it covers only a part of a layer you'll see a border where it cuts the model into pieces to have part of it printed with different settings.

amagr0, 2020-12-04:
Hello,
I'm doing a research project with metallic filaments, and one off the problems with the printed parts, is the air gaps between lines and layers, has you can see on this image.

[image]

I was looking for a solution, and it seams that the ironing feature could partialy resolve this, but ironing should occour on every single layer, I already tried modifie with other model has @Ghostkeeper sugested, but without success.

I send the Cura Project so you can see what could be wrong.
This parts must be solids, so the print strategy is to only print perimeters in a concentric way, to avoid rectractions as much as possible (doing with 100% infill and concentric it result on problematic issue, since the toolpath is not continuous).

[Metal_Solid_Parts.zip](https://github.com/Ultimaker/Cura/files/5644049/Metal_Solid_Parts.zip)

Thanks.

nubnubbud, 2021-01-26:
> Hello,
> I'm doing a research project with metallic filaments, and one off the problems with the printed parts, is the air gaps between lines and layers, has you can see on this image.
>
> I was looking [...]

please note that a number of people, myself included, have been concerned about that since ironing came out.:
> 
> https://github.com/Ultimaker/Cura/issues/5700
> https://github.com/Ultimaker/Cura/issues/3065
> https://github.com/Ultimaker/Cura/issues/5302
> https://github.com/Ultimaker/CuraEngine/issues/674
> https://community.ultimaker.com/topic/26593-please-include-an-iron-every-layer-option-for-clearer-prints/

but it really doesn't seem to be a concern. by writing toolpaths manually I've confirmed that the process can be used to make completely transparent parts. I'm pretty sure if you could do this with a metal-bearing filament it would be far stronger, as in order to be clear, the plastic needs time for the crystals to set and grow together.

https://imgur.com/gallery/hPk8Wl1 <images
https://www.reddit.com/r/3Dprinting/comments/jh5tva/refining_my_clear_petg_print_options_after_a/ <video

the piece shown is terrifyingly strong for how thin some of its parts are. indeed, most of my research went into eliminating those gaps, and increasing crystallization time without increasing print time.

Ghostkeeper, 2021-02-04:
I had actually done a similar experiment, but it produces a lot of overextrusion. This should be obvious, because the ironing adds 10% extra flow, so the object would become 110% dense and this material then extends towards the sides as the model is being printed.

Without ironing I've achieved similar results to you, by printing hotter (using PLA so it's a bit yellow-ish):
[image]

nubnubbud, 2021-02-05:
I used 1%-2% flow. an absolutely tiny amount that might still be too much, I'll explain why below. as well, I turn off part fans completely and 

if you look closely on mine, it gets foggy near the top. This has happened with all my PETG prints, and is directly related to the side the printer begins printing on. I think with many filaments, it's less about laying it down clear first, but returning it to the glass transition temperature and keeping it there with many nearby passes, to let it properly crystalize with the last layer.

this would explain why in most of my concentric prints at lower to average temperature, only about a centimeter in diameter is clear- without ironing, that's the only place the print head stays over and "cooks" for long enough to achieve the affect. 
[image]

simply put, I think it's less about pushing filament into little holes (which does help reduce artifacts a bit) but mostly about slowing the cooling of parts. you can do this two ways- by extruding hot enough to remelt previous layers, and by using the print head like... well, a clothes iron. 
each method has upsides and downsides, but it seems that the gist of it, is that ironing for clear parts takes far longer, but handles overhangs and complex shapes better, and is suited to temperature-sensitive filaments, while hotter extruding will suffer on overhangs and possibly be susceptible to oozing and other heat-related issues, but is much faster for flatter things.